IEE2463/classroom #3: "equivalencia" pines de los puertos Atmega con pines D0-D13 y A0-A5 de arduino en tinkercad

diegomarusic, 2020-03-18:
Hola, si yo quisiera usar el display de 7 segmentos en la Atmega328P lo que tendría que hacer es definir los pines de algún puerto como salida, por ejemplo el puerto C, escribir un código que haga lo que quiero y **conectar la Atmega 328P al display según las salidas que definí.** Sin embargo, no entiendo como hacer esta última parte de la conexión en el arduino UNO de tinkercad porque no entiendo cuál es la "equivalencia" de los pines de los puertos  de la Atmega328P con los pines D0-D13 y A0-A5 del arduino en tinkercad. ¿se entiende?

GutiJos, 2020-03-18:
Hola Diego!
Puedes buscar un pinout del arduino UNO y te aparecerá los pines correspondientes a la Atmega328p!

diegomarusic, 2020-03-18:
ok gracias

closed 2020-03-23 by GutiJos