IEE2463/classroom #6: Laboratorio 1 - Display de ánodo o cátodo común

moquezada, 2020-03-19:
Hola, ¿para el laboratorio 1 el display utilizado en el bonus debe ser de ánodo o cátodo común? Me fijé que en la laboratorio 2 piden explicitamente uno de ánodo común, pero para el bonus del laboratorio actual ya ocupé uno de cátodo común 🤔

GutiJos, 2020-03-19:
Dado que es tu bonus y eres libre de elegir cualquier elemento del programa para realizarlo, puedes utilizar el display de cátodo común. Asegúrate de utilizar el correcto en el Laboratorio 2!

Slds.

CamilaTurrieta, 2020-03-19:
Hola @moquezada! tal como dice @GutiJos si vas a ocupar el display para el bonus del lab 1, puedes elegir cual de ellos ocupar. De todas formas, la idea es que tengas claro cómo se usan ambos; por ejemplo: cómo funciona uno de ánodo común versus el de cátodo común. 

Saludos.

closed 2020-03-23 by GutiJos