IEE2463/classroom #7: Hora de simulador tinkercad

diegomarusic, 2020-03-19:


CamilaTurrieta, 2020-03-19:
Hola @diegomarusic ! Cómo primer error probable es que esté mal conectado. Si el error sigue, puedes subir tú código a tu repositorio de GitHub y una foto de la simulación idealmente, para que veamos si podemos darte ideas. Me cuentas!

diegomarusic, 2020-03-19:
subido!

CamilaTurrieta, 2020-03-19:
@diegomarusic ¿Subido en dónde?

diegomarusic, 2020-03-19:
IEE2463/laboratorio-01-diegomarusic Issues
https://github.com/IEE2463/laboratorio-01-diegomarusic/issues/1

te etiqueté nose si te llega la notificación

CamilaTurrieta, 2020-03-20:
@diegomarusic  efectivamente hay un problema al simular PWM con código C en tinkercad. Por el momento te sugiero que realices las ideas de bonus que son un poco más simples (es decir, que puedan ser escritas en C en tinkercad y funcionen). Mil disculpas por no avisar antes y estamos viendo como podemos solucionar esto. 

Ánimo y paciencia en estos tiempos de crisis.

diegomarusic, 2020-03-20:
oka no hay drama, gracias por responder todas mis preguntas jaja

CamilaTurrieta, 2020-03-20:
Estamos para eso 😄! gracias a ti por la motivación y preguntar ! ❤️

closed 2020-03-23 by GutiJos