IEE2463/classroom #8: ¿Es posible que el arduino de tinkercad no tenga la tierra de Atmega?

diegomarusic, 2020-03-19:
En todo los pinout que he encontrado la tierra está en un componente "sobre el arduino", sin embargo en tinkercad dicho componente no aparece. Entonces ¿con qué pin puedo usar la tierra de la atmega en el arduino de tinkercad?

moquezada, 2020-03-19:
No soy ayudante ni tampoco entiendo tanto tu pregunta lol Pero aquí te dejo un pin mapping del ATmega328P respecto a las salidas del arduino:

[image]

De todas formas, el Arduino de Tinkercad tiene como dos o tres pines de tierra disponibles, no sé porqué no podrías usar uno de esos 🤔

CamilaTurrieta, 2020-03-19:
Hola @diegomarusic tampoco entiendo muy bien tu pregunta :( .... ¿Podrías explicar más cuál es el problema? porque, tal como dijo @moquezada, no deberías tener problemas al usar las tierras que proporciona la placa de desarrollo del Arduino (que están unidas al atmega).

diegomarusic, 2020-03-19:
jajaj si gracias @moquezada, duda resuelta. Pensé que estaba conectando mal la tierra para medir una PWM con el osciloscopio, pero el problema es otro.

closed 2020-03-23 by GutiJos