IEE2463/classroom #15: Duda con respecto a tipos de variables

RafaelOrmazabal, 2020-03-20:
Estaba viendo un tutorial en youtube y en este se menciona que se recomienda declarar los números expresados en hexadecimal como una unsigned char variable y no como int. Saben porque es asi?

fffonceca, 2020-03-20:
Hola Rafa. El motivo principal es el uso y la cantidad de espacio que toman estas variables. Te recomiendo ver la cantidad de espacio que usa cada una de estas variables para que hagas la comparación justa, esto es importante, ya que las variables utilizan espacio en la RAM que es necesaria después. El tema de unsigned o signed es si es positivo o negativo, sin embargo, insisto en que veas documentación al respecto para que te quede más claro.

RafaelOrmazabal, 2020-03-20:
Muchas gracias, tienes alguna recomendación de donde partir investigando?

fffonceca, 2020-03-20:
Yo partí por esta página que ayuda bastante:
https://www.tutorialspoint.com/cprogramming/index.htm

closed 2020-03-31 by CamilaTurrieta