IEE2463/classroom #16: duda lab 2 "cuan sensible debe ser el botón"

diegomarusic, 2020-03-20:
Hola, me di cuenta que al hacer un click muy ligero sobre el botón, el código a veces reconoce el cambio de estado y otras veces no, por lo que quería saber si se considerará correcto para el task 1 del lab 2 mantener el botón apretado al menos medio segundo (en otras palabras hacer un click claro)

GutiJos, 2020-03-20:
Hola!

**NO** está permitido mantener el botón apretado para que se produzcan cambios en ciertas variables. El cambio se debe producir una vez que el botón es presionado.

Te recomiendo que leas toda la guía del laboratorio 2 y sepas qué es el efecto de bounce.

En conclusión, deberás implementar un método de debounce por software, es decir, un código que te permita la correcta lectura del botón. Hay mucha información disponible en internet y tengo entendido que en Canvas hay una guía de AVR en donde se explica cómo hacer debounce. Recuerden que es importante que entiendan el código en su totalidad y pongan las referencias si corresponde.

Espero haberte ayudado.
Saludos.

diegomarusic, 2020-03-20:
dale gracias, ese es mi problema. Está cambiando cuando lo suelto y no cuando lo apreto. Veré que hacer!

RafaelOrmazabal, 2020-03-20:
Donde esta esa guia?

closed 2020-03-31 by CamilaTurrieta