IEE2463/classroom #17: Duda con respecto a declaracion de inputs y outputs

RafaelOrmazabal, 2020-03-20:
Hola, estaba trabajando en un circuito en tinkercad y estaba declarando primero que puertos eran outputs y luego que puertos eran inputs. No obstante, no me estaba funcionando el circuito. Ahora bien, cambie el orden y declare primero los puertos que eran inputs y funciono perfecto, saben porque podría ser?

jdiaztoledo, 2020-03-20:
Hola!

Podrías enviar como defines los inputs y outputs en tu código? Se me ocurre que quizás puedas estar definiendo los puertos de salida con una igualdad en vez de con un or bit a bit, es decir, por ejemplo lo estás haciendo así:

DDRD = 0b00000001;

En vez de:

DDRD |= 0b00000001; 

Al hacerlo de la primera forma estás sobreescribiendo todos los registros que anteriormente ya habías definido, en cambio de la segunda forma cambias solo el bit que especifica. Es por eso que quizás al definir los inputs después, estás "borrando" los registros ya definidos como output.

Espero que sea eso, sino de todas formas podemos seguir ayudándote.

Saludos :)

RafaelOrmazabal, 2020-03-20:
Hola, gracias por la ayuda. Asi lo hice inicialmente (no me funciono):
  DDRD |= (1<<PD1);		
  DDRD &= !(1<<PD2);
Y asi lo hice después cuando me funciono:
DDRD &= !(1<<PD2);
  DDRD |= (1<<PD1);

porellanau, 2020-03-20:
Hola!

Te recomiendo revisar el Datasheet del ATmega328, en especifico la sección de I/O Ports, ya que estos registros tienen valores por default :eyes:

RafaelOrmazabal, 2020-03-21:
Hola, vi la hoja de datos y busque información en internet y sigo sin entender el problema. Además, encontré un video un youtube de alguien que estaba haciendo exactamente el mismo circuito que yo y definía primero los outputs y luego los inputs (solo que el usaba el puerto B para ambos, en cambio yo use el puerto D) y le funcionaba perfecto. Además, sabiendo cual es el valor que tienen los registros por default (viendolo en la hoja de datos), hice las operaciones bitewise que mencioné arriba a mano, e independiente si pongo el input o el output primero, el valor de los registros de D son los que quiero, pero no funciona si declaro primero el output. 

Finalmente, intente de ver con un multímetro si había otro problema y me di cuenta que cuando declaro primero el output y luego el input, el pin que debería entregarme 0 o 5V, es capaz de entregarme 5 volts operando como high, pero cuando es low me entrega 3.66V. En cambio, cuando declaro el input primero, el pin que funciona como output funciona perfecto, entregandome 0 o 5V. Esto último me hace pensar que el pin en ambos casos si tiene la capacidad de funcionar como output, ya que en ambos casos puede entregar un voltaje high y un low, solo que en un caso no puede fijarse en 0V. Además me indica que el pin de input en ambos casos funciona bien, puesto que cuando apreto el pulsado en ambos casos el output baja su voltaje, solo que en uno no baja al voltaje necesario (baja a 3.66 en vez de 0V). Les agredecería mucho si me pueden ayudar con este problema, ya que me interesa aprender que estoy haciendo mal o si hay algún tipo de bug con tinkercad.

fffonceca, 2020-03-21:
Para hacer un análisis más a fondo podrías enviarnos tu código directamente a los ayudantes para poder ayudarte de mejor forma, ya que es muy probable que la definición de registros no esté totalmente bien acabada. Como dijo @porellanau hay registros que se definen por default, además de otros registros aparte de DDRX que estén provocando el problema.

RafaelOrmazabal, 2020-03-21:
Ok, muchas gracias les voy a mandar por correo mi proyecto de tinkercad.

closed 2020-03-31 by CamilaTurrieta