IEE2463/classroom #18: Tutorial AVR

cefernandez1, 2020-03-20:
Hola, donde se encuentra el tutorial AVR(por Marcelo Guarini)?. Gracias de antemano.

porellanau, 2020-03-20:
Hola!

El tutorial se encuentra en Canvas, en la sección de archivos.

Saludos!

RafaelOrmazabal, 2020-03-20:
[image]

RafaelOrmazabal, 2020-03-20:
No veo la sección de archivos

mfbarahona, 2020-03-20:
Hola, revisa el Material Complementario en Módulos.

Saludos

Jbijit, 2020-03-20:
Hola, en que parte? lo acabo de buscar y no lo encuentro. No hay ningún archivo con el nombre de tutorial AVR.

Saludos.

mfbarahona, 2020-03-20:
@Jbijit GuiaAVR = Tutorial AVR

Saludos

Jbijit, 2020-03-20:
@mfbarahona 

Tengo abierto Canvas y en la sección de módulos y no hay ningún archivo con ninguno de los 2 nombres.

Saludos

porellanau, 2020-03-20:
Hola!

¿Podrías mandarme un pantallazo de la sección de módulos? Está al final! :smiley:

Saludos!

Jbijit, 2020-03-21:
Acabo de actualizar la pagina de nuevo y me aparecieron los nuevos archivos!

Muchas gracias!!
Saludos!

closed 2020-03-31 by CamilaTurrieta