IEE2463/classroom #19: Problema Led

Jbijit, 2020-03-20:
Estimados,

Tengo un problema con el "funcionamiento del Led", mi código prende y apaga el Led pero este pasa por una fase intermedia en la cual queda prendido con menos intensidad, y no entiendo por que. Es como si el circuito se cargará durante el proceso.......

Quedo atento a sus comentarios, muchas gracias.
Saludos.

porellanau, 2020-03-21:
Hola!

Muy probablemente el problema sea por la simulación y no tu culpa ni la del programa. Si quieres quedarte más tranquilo podrías hacer una issue privada y compartir el link de tu simulación. Ahí puedo confirmarte si es normal o no :wink:

Saludos!

Jbijit, 2020-03-21:
Sería ideal! Muchas gracias!

Saludos!

closed 2020-03-31 by CamilaTurrieta