IEE2463/classroom #21: Problema con la simulación

cefernandez1, 2020-03-21:
Hola,

Al simular en tinkercad un código que me permite generar una PWM, me pasa que el tiempo de simulación no corre. Al buscar en internet encontré que podía ser útil el uso de retrasos en el código, pero no funcionó. ¿Hay algún otro tipo de solución para este problema?

Gracias de antemano.

fjhuanca, 2020-03-21:
Hola @cefernandez1 el tiempo no corre en absoluto, o avanza muy lento? Podrías crear una issue privada en tu repositorio con el código para echarle un vistazo. Saludos

diegomarusic, 2020-03-21:
Hola, no soy ayudante, pero me dijeron que al parecer tinkercad tiene problemas para trabajar con PWM. Te recomiendo tratar de hacer otro bonus (yo igual hice una que debería haber funcionado y me pasaba el mismo problema).

closed 2020-03-31 by CamilaTurrieta