IEE2463/classroom #22: Duda en bonus laboratorio 2

sebapulgar, 2020-03-21:
Hola muy buenas esperando que estén muy bien quería saber si para el bono que pide "Guardar cantidad de veces que se presionó el botón, luego el Led parpadea esta misma cantidad de veces", es posible utilizar un pulsador externo al presente en la placa, para comenzar el parpadeo del led, ¿o solo puedo usar el pulsador reset presente en la placa para todo el proceso?
mil gracias

jdiaztoledo, 2020-03-21:
Hola!

Efectivamente para ese bonus debes utilizar un pulsador externo ya que, como dices, el botón de la placa solo sirve como reset.

Saludos:)

sebapulgar, 2020-03-21:
Gracias!

mlibarra, 2020-03-22:
Hola, 
tengo la misma duda y no me quedó claro. Se pueden usar dos pulsadores? uno para contar cantidad de veces que se presionó el botón, y otro para comenzar el parpadeo del led?

GutiJos, 2020-03-22:
Hola @mlibarra !
La dinámica del bonus es que ustedes expresen alguna funcionalidad creativa. Lo que aparece en la rubrica es sólo una guía. Recuerden que el bonus no es un task con el que deben cumplir ciertos requerimientos, por lo tanto ustedes son libres de realizar cualquier funcionalidad y de implementarlas como quieran.

A lo que iba @jdiaztoledo es que el pulsador que viene en la tarjeta de Arduino es un botón de reset y no te servirá para para utilizarlo como un input de tu programa. Por lo tanto, debes usar algún botón externo para que puedas interpretarlo como un input a tu circuito.

Espero te haya quedado un poco más claro.
Saludos.

mlibarra, 2020-03-22:
Gracias! entendi todo,
Saludos

felipesanchezvaras, 2020-03-22:
@mlibarra Favor revisar el [comentario](https://github.com/IEE2463/classroom/issues/22#issuecomment-602135122) anterior para evitar confusiones. 

Saludos ! :)

closed 2020-03-31 by CamilaTurrieta