IEE2463/classroom #23: PWM en Bonus (prohibición por el momento)

fjhuanca, 2020-03-21:
Hola a todos! Como equipo de ayudantes hemos detectado un bug importante en Tinkercad a la hora de simular PWMs, por lo tanto se nos hace inviable poder evaluar cualquier bonus que la incluya en su funcionamiento.

Por lo tanto **se les solicita que no incurran en uso de PWMs en sus bonus** al menos por ahora, pero dado que no es un bug confirmado, si alguien tiene una solución o a hecho algo que le da como resultado una simulación exitosa les pido nos lo comuniquen para poder reconsiderar la situación a futuro.

Les pido disculpas por los inconvenientes, estamos trabajando para que los laboratorios sean lo más amenos posible pero hay que considerar que es un semestre bastante particular.

Éxito, ánimo y cuídense!

cefernandez1, 2020-03-22:
Hola ¿Si pude generar PWM sólo a partir de timers, puedo entregar de todos modos el bonus de PWM?

CamilaTurrieta, 2020-03-22:
@cefernandez1 para estar más segura de exactamente a que te refieres y poder responderte mejor; ¿podrías comparte el código y/o el link de la simulación en tu repositorio privado de github?

CristobalMery, 2020-03-22:
Hola! Yo también logré generar una PWM a partir del uso de comparación de los timers. ¿El bonus aplicará?

felipesanchezvaras, 2020-03-22:
@CristobalMery Misma situación, ¿podrías comparte el código y/o el link de la simulación en tu repositorio privado de github?

CristobalMery, 2020-03-22:
Acabo de subir mi código de la PWM a mi repositorio privado. En este se encuentra un comentario con el link al circuito de tinkercard.
Saludos!

felipesanchezvaras, 2020-03-22:
@CristobalMery @cefernandez1 Autorizados para presentar la PWM para mañana, si alguien más lo logra hacer, también será considerado.

A los demás. Buscaremos una forma de darle lineamientos para el futuro puedan implementar también PWM en TinkerCad.

Saludos !

closed 2020-03-31 by CamilaTurrieta