IEE2463/classroom #42: Duda botón

JavieraSanMartin, 2020-03-27:
Hola; Espero que estén bien. Les escribo porque estoy perdidísima con lo del botón, no se si podrían recomendarme algún lugar donde investigar o algo similar, porque realmente estoy muy desorientada. No se cómo hacer una especie de DigitalRead para el pin que elegí como entrada. Me quedo atenta.

Gracias

GutiJos, 2020-03-27:
Hola @JavieraSanMartin !
Te recomiendo leer la GuíaAVR que se encuentra en Módulos>Material complementario>GuíaAVR. En el título 6 se refiere a las entradas y salidas digitales, y en específico en la página 15 hay un código tipo que te sirve para configurar el botón como input y leer cuando es presionado.

Saludos y ánimo.

fjhuanca, 2020-03-27:
@JavieraSanMartin Hola! acá te dejo un video que encontré en español por si te encuentras aún muy perdida: [Video AVR Pines](https://www.youtube.com/watch?v=1WdtQHyoT3A), en inglés puedes encontrar mucha información buscando como "AVR input output setup C" o similares. Cualquier cosa no dudes en preguntar!

Éxito y saludos!

JavieraSanMartin, 2020-03-28:
Muchas gracias a ambos :D

closed 2020-03-31 by CamilaTurrieta