IEE2463/classroom #43: fotorresistor

diegomarusic, 2020-03-27:
Si lo conecto como aparece en la imagen, al variar la luz del fotorresistor cambia solo su resistencia pero no su voltaje o corriente. Sin embargo, si lo conecto a una fuente externa mediante una resistencia logro que cambie su voltaje en 0.54V, pero me aparece "error" al medir la resistencia.

mi duda es ¿el ADC debo hacerlo con una medición de cambio en la resistencia o en el voltaje o yo elijo?

saludos

fjhuanca, 2020-03-27:
Hola @diegomarusic, mira si entiendo bien tu pregunta, la respuesta es que solamente puedes hacerlo mediante un cambio en voltaje. No tienes como medir un cambio en la resistencia del fotoresistor con el microcontrolador (al menos directamente). Ojo que debes considerar que el MSP funciona con 3.3V para tus cálculos.

Éxito!
Saludos

diegomarusic, 2020-03-27:
solved

closed 2020-03-31 by CamilaTurrieta