IEE2463/classroom #46: Problema Tinkercad

Jbijit, 2020-03-28:
Estimados, 

Nuevamente me ocurre que los LEDs (ahora del display) se "cargan" y se demoran en "descargar", lo que no me permite aumentar la  velocidad de cambio de los números, ya que estos no se verían bien . Lo más raro de todo, es que si fuera problema de código estos se mantendrían bien prendidos (pero no lo hacen), sino que se van atenuando de a poco hasta apagarse.

¿Que se puede hacer en este caso?
Al parecer es como una especie de problema con el simulador y mi conexión a este o no se, por lo que de ser así, ¿abría alguna penalización por que el cambio de los números se vea?

Saludos.
Joaquín.

CamilaTurrieta, 2020-03-29:
Hola, si quieres puedes compartir una issue en tu repositorio privado y así podemos entender más lo que nos cuentas. ... por mientras, cuando te ocurre el problema ¿el reloj de simulación del tinkercad se detiene o avanza extraño? 

Me cuentas!

Jbijit, 2020-03-29:
Listo, acabo de hacer la issue en mi repositorio privado.

Muchas gracias.

Saludos.

closed 2020-03-31 by CamilaTurrieta