IEE2463/classroom #53: Objetivos Lab 3 (Duda Enunciado muy general)

jffarren, 2020-04-02:
No entiendo bien qué es lo que tenemos que hacer exactamente en este laboratorio. No me queda claro en qué punto se une el fotoresistor con el led y qué debemos hacer con el código entregado.  Mis disculpas por publicar esto tan cercano a la fecha de entrega, pero por varios motivos no he podido. ¡Gracias y quedo atento!

GutiJos, 2020-04-02:
Hola @jffarren 
El laboratorio trata de variar la intensidad de un LED mediante una PWM, y esta PWM debe ser variable de acuerdo a la cantidad de luz que recibe tu fotoresistor.
Entonces tendrás dos grandes etapas. La primera es leer estas variaciones que se generan en tu fotoresistor mediante un ADC. La segunda trata de convertir esas variaciones y convertirlas en una PWM (variable) que alimentará tu LED. 
A modo de ejemplo, si tu fotoresistor está completamente iluminado, se debe reflejar en una PWM con duty cycle de 100%, si la iluminación es de un 50%, el duty cycle debería ser de un 50% y así.

Espero te haya quedado un poco más claro.

Saludos.

mlibarra, 2020-04-02:
Hola, 
Según entendí, sólo debemos variar el duty cycle de la PWM que ya nos dan?

GutiJos, 2020-04-02:
Lectura de las variaciones del fotoresistor mediante un ADC y reflejarlo en una PWM con duty cycle variable.

Saludos.

closed 2020-05-19 by felipesanchezvaras