IEE2463/classroom #56: Bonus UART

cefernandez1, 2020-04-03:
Hola, al hacer el bonus que consiste en controlar la PWM con UART me pasa que la tarjeta que recibe la información nunca entra en la interrupción que corresponde a la recepción de información, mientras que la tarjeta que envía la información sí pasa por la interrupción correspondiente. No sé si corresponde a un problema de mi código o tinkercad. Gracias de antemano.

CristobalMery, 2020-04-03:
Hola Cristian. Tengo exactamente el mismo problema. La tarjeta que está recibiendo los datos nunca entra a la interrupción que debería entrar cuando los recibe.

GutiJos, 2020-04-03:
Hola @cefernandez1 y @CristobalMery 
Para ambos: están usando una tarjeta de arduino para enviar info y otra para recibir la info y generar la PWM?

Otra cosa: cómo están verificando si efectivamente envían la señal o la reciben? Por el monitor serial?

Saludos.

cefernandez1, 2020-04-03:
Sí, pero no estoy usando el monitor serial, sólo prendo un led en caso de que se entre a la interrupción que corresponde a la recepción y el envío de datos.

CristobalMery, 2020-04-03:
Hola!
Si, estoy ocupando una tarjeta de arduino para enviar la info y otra tarjeta para recibir la información y generar la PWM.

GutiJos, 2020-04-03:
@cefernandez1 y @CristobalMery 

Primero les explico que lo que están tratando de hacer es parte de la materia en los siguientes 2-3 laboratorios, donde se ve la transmisión y recepción de datos y además se ven las interrupciones de UART. Para que lo tengan en cuenta.
Además les quería comentar que el control por UART lo podrían hacer utilizando sólo un microcontrolador y enviar los datos a través del monitorial serial al arduino.

Ahora sobre su implementación, les recomiendo que verifiquen por separado si su código es capaz de enviar y recibir caracteres con ayuda del monitor serial. Y luego, prueben las interrupciones de transmisión y recepción. Por lo que entendí arriba, solamente la recepción no les funciona, traten de probar ese código por separado enviándole datos a arduino mediante el monitor serial.

Saludos.

closed 2020-05-19 by felipesanchezvaras