IEE2463/classroom #60: Posible bug en Tinkercad con UART

tito21, 2020-04-03:
Intentando el bonus de enviar la luz medida por UART encontré un comportamiento extraño en el simulador.

Dentro de la función main tengo un loop infinito en donde constantemente leo el valor del ADC y después lo envió (junto con un mensaje de texto) por UART. Este loop se corta y no se envían los mensajes ni se lee el ADC. Por otro lado el mensaje SI se envía cuando está dentro de una interrupción de botón. Probando descubrí que el problema solo ocurre cuando llamo a la función que me lee el  ADC.

Sospecho de un bug en el simulador porque cuando cargo el mismo programa en un Arduino UNO físico el programa funciona como esperaba. Se envía el mensaje continuamente y con la interrupcion de botón.

fffonceca, 2020-04-03:
Si lo probaste en el físico y te funciona claramente es un problema de simulación. De todas formas, podrías mandar tu código para hacer un análisis más profundo de este y poder darte un feedback de como optimizarlo. Optimizar en el sentido de aprovechar bien los tiempos de interrupción con el envió por UART para evitar problemas.

tito21, 2020-04-04:
En mi repo privada deje el link

RafaelOrmazabal, 2020-04-06:
@tito21 Como hiciste para cargar el programa en C en un Arduino UNO, usas Arduino IDE o usas otro software?

tito21, 2020-04-06:
Usando la IDE de Arduino.  Intenté usar Atmel studio pero no me funciona.

closed 2020-05-19 by felipesanchezvaras