IEE2463/classroom #62: Entrega circuito tinkercad

cefernandez1, 2020-04-05:
Hola, no veo en canvas ningún link para compartir el circuito de tinkercad ¿ Cuándo va a estar disponible para subir la tarea?
Gracias de antemano.

felipesanchezvaras, 2020-04-05:
Hola ! 

Como fue informado en el último anuncio de Canvas, debes compartir tu link del circuito de TinkerCad en el Readme de tu repositorio, junto con una captura de pantalla de tu circuito.

Saludos !

assilva2, 2020-04-05:
y como ponemos una imagen en el codigo que subimos? traté con copy y paste, pero no funcionó.

moquezada, 2020-04-05:
Con una rápida búsqueda en google encontré [esto](https://markdown.es/sintaxis-markdown/#imagenes). En primer lugar, puedes subir las imagenes a tu repositorio y luego utilizar el link de esas imágenes para referenciarlas en formato markdown :eyes:

closed 2020-05-19 by felipesanchezvaras