IEE2463/classroom #63: LAB 04 - Dudas de enunciado

felipesanchezvaras, 2020-05-19:
Espacio para consultar todo tipo de dudas de enunciado

dabalada, 2020-05-22:
Respecto a la sección 5 del enunciado.
Cuales serian los requerimientos mínimos en caso de usar linux (mas concretamente ubuntu)? O requiero obligatoriamente correr windows en una maquina virtual?

PD: No es necesario un tutorial con todos los comandos, con que me digan que instalar par buscar como hacerlo esta bien.

felipesanchezvaras, 2020-05-22:
@dabalada 

No es necesario que utilices una máquina virtual de Windows, la interfaz está basada en una realizada en Linux, por lo que deberías ser capaz de realizarla ahí.

En particular necesitas contar con Glade y con GTK+ 3 Development Library. En la siguiente página https://prognotes.net/2015/12/installing-gtk-3-and-glade-development-tools-in-linux/ está un tutorial de cómo hacerlo en Linux. Este es el mismo tutorial que se utilizó para realizar la interfaz del juego. [GTK Glade C Programming Tutorial](https://prognotes.net/gtk-glade-c-programming/), así que los créditos van para la página de [Programmer's Notes](https://prognotes.net/).

Ojo que hay que hacer unos pequeños cambios en el código. 

En el archivo main.c debes comentar la línea 22 que contiene la biblioteca `#include <windows.h>`.

Por otra parte, al archivo compile.script se le debe agregar `-export-dynamic` al final. Quedando como:

`gcc -Wno-format -o main-bin src/game_logic.c src/main.c -Wno-deprecated-declarations -Wno-format-security -lm `pkg-config --cflags --libs gtk+-3.0` -export-dynamic`

CristobalMery, 2020-05-24:
Buenas noches. Estoy un poco confundido respecto de cual debe ser la funcionalidad del juego en sus diferentes modos.

**Respecto del  modo LinearBuffer:** El juego debe lanzar la secuencia de n elementos apenas se comienza? y el jugador intentar repetirla? O bien; hay que ir mostrando primero 1 número, luego 2 números, y asi sucesivamente hasta que se alcancen los n elementos?

**Respecto del modo RingBuffer:** No entiendo que diferencia tendría para el jugador este modo respecto del linear buffer. O bien la diferencia es únicamente en el código?

felipesanchezvaras, 2020-05-24:
@CristobalMery 
Respecto del modo LinearBuffer:
El juego debe mostrar primero 1 numero, luego el jugador presiona 1 colores.
Luego muestra 2 (repitiendo el anterior), luego el jugador presiona 2 colores.
Luego muestra 3 (repitiendo los 2 anteriores), luego el jugador presiona los 3 colores. 
Y así sucesivamente, el siguiente video es muy ejemplificador a qué es lo que se pide.

https://www.youtube.com/watch?v=vLi1qnRmpe4

La diferencia cae en que si , por ejemplo n = 5, el número máximo de colores a mostrar será 5 y el juego terminará, por lo que no podrá seguir aumentando su puntaje.

Respecto del modo RingBuffer: A diferencia del anterior, en este hay que recordar los últimos n elementos. Si comenzamos el juego, la secuencia sería la siguiente (con números representaré el color que aparecerán

| Secuencia a mostrar | Lo que debe presionar el juego | 
|  1  |   1  |
|  1  |   1  |
|  1  |   1  |
|  1  |   1  |
|  1  |   1  |
|  1  |   1  |
|  1  |   1  |
|  1  |   1  |

CristobalMery, 2020-05-24:
Perfecto! Ahí me quedó clarísimo.
Muchas gracias!

mlibarra, 2020-05-25:
Hola, 
Estoy confundido con la separación entre main y game_logic. En el video se menciona que para modificar la animacion de las imagenes se debe modificar la función "show_mem_function" que está en game_logic, pero en el archivo main al comienzo está "g_snprintf(widgets->game_show,10, "%s", "023130132");" que genera la secuencia de las imágenes. Entonces si quiero mostrar una nueva secuencia, tendría que modificar main, game_logic o ambos? Lo que entendí es que el main es el que conecta con la interfaz, por ende debería ser ahí. Pero también señalan que la lógica del juego va en game_logic

mlibarra, 2020-05-27:
Hola! El score del juego en qué momento debe aumentarse? Tengo la duda si es en cada color "bueno" que apriete el jugador, o al final de cada ciclo si presionó todos los colores bien, es decir, si por ejemplo sale la secuencia 1-2-3, y el jugador presiona 1-2-3, son 3 puntos o sólo 1? Saludos!

felipesanchezvaras, 2020-05-27:
@mlibarra 

El score debería cambiar al momento de completar una secuencia correcta.

1-2-3 (3 pts)
Siguiente ronda
1-2-3-4 (aumenta a 4 pts cuando completa)
Siguiente ronda
1-2-X (Si se equivocó en este punto, el jugador queda con 4 pts, ya que fue su última puntuación máxima y no 2, que sería la puntuación si se fuese activando cada vez que se aumenta).

mlibarra, 2020-05-27:
@felipesanchezvaras Gracias! Ahí me quedó claro. Tengo dos dudas más.  ¿Se pueden añadir funciones a game_logic.h? Por ejemplo añadir la función "update_score" para poder usarla en el game_logic.c además del main.c (donde ya está), o se debe utilizar sólo en el main.c?
Mi segunda duda es que no comprendo como funcionan estas lineas del game_logic.c:

"// Returns the state of the game as either busy
game_state get_game_state(app_widgets *app_wdgts, gchar button)
{
    // Determina si el estado debe cambiar de BUSY -> SHOW -> END
}"

Se está creando una función tipo estructura? algo así entendí. 
Gracias de antemano!

felipesanchezvaras, 2020-05-30:
@mlibarra 
Lo ideal es que update_score se mantenga en el main.c, ya que así se realiza una mejor separación entre lo que es interfaz y lo que es la lógica del juego. Pero eso no quita que sí pueda ser realizado y utilizado desde game_logic.c

En el archivo game_logic.h tu declaras los prototipos de la funciones, por lo que debes tener ojo qué defines aquí.

```
// Returns the state of the game as either busy
game_state get_game_state(app_widgets *app_wdgts, gchar button)
{
// Determina si el estado debe cambiar de BUSY -> SHOW -> END
}
```

Esta es la declaración de una función, que podría ser utilizada, pero no necesariamente lo será. Lo que está diciendo es que el tipo de variable que retorna es `game_state`

PamelaRomero, 2020-06-03:
Buenas noches, quería saber si es posible que me puedan guiar/explicar un poco sobre como establecer la secuencia a memorizar para que sea aleatoria. No entiendo muy bien si debería hacer una función totalmente a parte para que cree la secuencia e integrarla en el programa o hay que utilizar una función ya presente en el programa main.c . 

Muchas gracias de antemano c:

CamilaTurrieta, 2020-06-03:
Hola @PamelaRomero para generar la secuencia aleatoria no hay ninguna función que ya este originalmente en el main o en la librería. Por lo que tu debes implementar, en alguna parte de la lógica del juego, que se vaya creando la secuencia aleatoria. La decisión de cómo hacerlo es tuya. 

Por ejemplo, lo que mencionas de generar una función aparte que haga eso es una buena idea. Así cada vez que quieras crear alguna secuencia aleatoria llamas a esa función.  

Entonces, es importante que te fijes en donde debes ir cargando los datos para que estos sean los que se muestran en la interfaz. Luego, ver cómo asignas a eso elementos aleatorios. Que esto último sería por ejemplo a través de una función que devuelva una cantidad n de elementos aleatorios. 

¿Me explico? ¿Te surgen más dudas con esto? Me cuentas!

PamelaRomero, 2020-06-03:
Yaa entiendo!! Muchaas gracias por la respuesta!!! c:

Jbijit, 2020-06-05:
Hola, 

Tengo 2 dudas de enunciado.
1) Los puntajes aumentan en 1 punto por ronda o bien por la cantidad de botones apretados en la ronda, explico diferencia (vi la issue anterior de esto y me quedo la duda):

Caso1 (1 punto por ronda completada):
Score=0 -> score = 1 -> score = 2 -> score = 3

Caso 2 (se suman la cantidad de imágenes presionadas al puntaje total):
Score=0 -> score = 1 -> score = 3 -> score = 6.

2) Cuando se dice que se ignora el cambio de modo, hay que impedir que cambie el botón en la interfaz o bien se puede hacer que cambie el botón pero no el modo de juego, adjunto imagen explicando el punto final:

[image]

Muchas gracias.

felipesanchezvaras, 2020-06-05:
@Jbijit Cualquiera de las formas sería valida. Lo importante es que el cambio de modo de juego solo pueda ocurrir en GM_INIT o en GM_END. 
 
Una vez se esté en GM_SHOW o GM_BUSY no se puede cambiar el modo de juego, dado que ya fue determinado al comienzo.

Jbijit, 2020-06-05:
@felipesanchezvaras Perfecto, muchas gracias!!

Jbijit, 2020-06-06:
Hola, tengo una duda del informe que piden, ¿que tan detallada tiene que ser la explicación que se pide y tiene que tener algún formato en particular?

dabalada, 2020-06-06:
Sumándome a la issue anterior, en el informe tengo que explicar las funciones o solo menciona que cosas implementé y que cosas no. Si tengo que explicar la funciones, no me cabe en solo 2 paginas.

CamilaTurrieta, 2020-06-06:
Hola! @Jbijit y @dabalada. No debe ser tan detallado el informe. No es necesario explicar todo. La idea es que sea una guía que facilite la revisión de su programa. La idea es que sea un mini resumen de lo qué hicieron, qué no hicieron y que no resultó o algún adicional que crean que es importante comentar.  

En concreto:
1.  No debe ser una explicación muy detallada. La idea es que contenga: 
          -  El resumen de que se realizó. 
          -  Mencionar si se uso librerías adicionales u algo adicional. 
          -  Resumen de su solución implementada. 
  
2.  No deben mencionar la explicación de todas las funciones porque la idea es que eso se encuentre en el código. 

3. El formato de escritura no debe ser nada especial, tu elección mientras sea ordenado. La idea es que alguien más lo pueda entender. Lo importante: respetar el número de páginas. enviar el archivo en pdf y su nombre. 

Espero haya quedado más claro. Si no, me cuentan!

Michel1297, 2020-06-08:
Hola, respecto al bonus de dificultad, se debe poner un boton que cambie la dificultad o esta debe ir subiendo a medida que se avanza en cualquier juego? Saludos

felipesanchezvaras, 2020-06-08:
@Michel1297 Queda a tu decisión. O puedes implementar ambos. Mientras esté especificado en el informe cómo lo implementaste, está ok!

closed 2020-06-16 by felipesanchezvaras