IEE2463/classroom #64: LAB 04 - Dudas de Interfaz Gráfica

felipesanchezvaras, 2020-05-19:
Espacio para consultar todo tipo de dudas relacionadas con la Interfaz Gráfica

gonzalezconstanza, 2020-05-21:
Hola, estoy perdida en cómo programar la interfaz gráfica...ya seguí los pasas de requerimiento mínimo para ejecutar, pero no entiendo como abrir la interfaz.

felipesanchezvaras, 2020-05-21:
Hola @gonzalezconstanza , 

Depende del sistema operativo que tengas, si me puedes indicar cuál, te podría guiar de mejor forma. En mi caso trabajo con Windows y para abrir la interfaz debo presionar el archivo `main-bin.exe`

[image]

Otra opción es abrir la consola de MSYS, ir a la carpeta de tu programa utilizando el comando `cd` y ahí abrir el archivo generado con la compilación.

Quedo atento.

mlibarra, 2020-05-22:
Hola, cuando intento iniciar el main-bin.exe me dice que la aplicación no pudo iniciar correctamente. Antes que esto me decia que no encontraba el archivo libobject-2.0.0.dll, libglib-2.0-0.dll y libgtk-3-0.dll, leí que se podían descargar y colocarlos en la carpeta System32, pero luego de eso me el main-bin me arroja "no se pudo iniciar"

dabalada, 2020-05-22:
Hola, tras compilar y ejecutar el binario, se abre la interfaz gráfica pero no se ven las imágenes en los botones 
[image]

PD: El sistema operativo es Ubuntu 20.04

felipesanchezvaras, 2020-05-22:
@dabalada Puedes probar presionando el botón start ? posterior a eso, deberías ser capaz de presionar las imágenes. También podrías agregar una captura de pantalla de lo que aparece en consola (puede que ahí aparezca el error).

dabalada, 2020-05-22:
@felipesanchezvaras Al presionar start no pasa nada. Adjunto captura de consola.
[image]

felipesanchezvaras, 2020-05-22:
@mlibarra Posiblemente eso ocurra porque no tienes las variables de entorno en tu PATH. En el video que subí, en particular a partir del [minuto 2:53](https://youtu.be/pDlryWL4YF8?t=173) indico lo que se debe configurar.

felipesanchezvaras, 2020-05-22:
@dabalada ¿ Puedes revisar que las imágenes se encuentren en la carpeta que aparece ahí? Por otro lado, en el código, archivo main.c, comenta todas las lineas que parten con `G_MODULE_EXPORT ` Ya que eso es necesario para Windows.

dabalada, 2020-05-22:
@felipesanchezvaras Comente todas las lineas que me dices y revise que las imágenes estuvieran en la carpeta correspondiente, sigue sin mostrarme las imágenes en los botones

Jbijit, 2020-05-22:
Hola, 
Trato de descargar el MSYS2 y cuando lo ejecuto y corro la primera linea que dice la página del programa (pacman -Syu), llego a esto:

[image]

El tema es que si lo cierro no lo puedo volver a abrir, como que hace que se va a abrir y luego se cierra inmediatamente, por favor si me pudieran ayudar sería ideal, llevo desde ayer con esto. 

Además, se queda pegado ahí, por lo que no puedo correr las otras lineas que se piden en particular para la tarea.

Saludos.

mlibarra, 2020-05-22:
@felipesanchezvaras seguí los pasos del video. Tengo en el path ambas direcciones pedidas, además de haber instalado todo (también verifiqué que está gcc). Al iniciar con la consola al archivo compile.script me arroja este error:
 
[image]

felipesanchezvaras, 2020-05-23:
@Jbijit Podrías intentar abrir el instalador con permisos de administrador, tal vez eso soluciona el problema. Por otra parte veo que la actualización de los paquetes fue realizada. Prueba si es que puedes ejecutar los siguientes pasos de la página https://www.gtk.org/docs/installations/windows/

En particular 
`pacman -S mingw-w64-x86_64-gtk3`

@mlibarra ¿Eso lo estás ejecutando en una ventana de comandos de Windows ?¿ o en MSYS ?  ya que debe ser realizado en el segundo para que funcione. Otra opción es fijarte que el PATH tenga las variables de entorno para todos los usuarios y no solo el usuario actual.

Además podrías probar ejecutando gcc en MSYS y ver que te aparece.

mlibarra, 2020-05-23:
@felipesanchezvaras lo estoy ejecutando en MSYS de 64, y cuando ejecuto gcc dice "bash: gcc: no se encontró la orden". Pero si ejecuto gcc -v en cmd si me indica la versión como aparece en el video. Quizás la consola no está conectando con gcc

mlibarra, 2020-05-23:
lo raro es que si abro la consola de gitbash, si encuentra gcc (si pongo gcc --version), e incluso llega a tratar de abrir el archivo compile.script pero arroja error. La consola de gitbash a diferencia de la de Msys dice "master"

Jbijit, 2020-05-23:
@felipesanchezvaras ese es el tema, al cerrar esa ventana no la puedo volver a abrir, no se porque y por lo mismo no puedo correr esas lineas que dice esa pagina. Se queda pegado en ese punto y si lo saco después dice que te interrumpió la conexión.

mlibarra, 2020-05-23:
@felipesanchezvaras logré abrir bien el archivo main-bin.exe, y para que la consola reconociera gcc tuve que instalar "pacman -S mingw-w64-x86_64-toolchain" lo cual encontré en "https://github.com/msys2/MSYS2-packages/issues/293". De esta forma al ejecutar el compile.script no tira error y me reconoce la version 10 de gcc (el cmd me reconoce 9.2). Estará bien? entendí que el compile.script se utiliza para compilar, y después al ejecutar la consola deberían verse lo cambios, o no?

felipesanchezvaras, 2020-05-23:
@mlibarra Justo estaba respondiendo referenciando una Issue similar a la que pusiste. Gracias por el aviso. Lo que dices tienes razón. Para ver los cambios en el codigo, debes compilar cada vez con el archivo compile.script (o escribiendo en consola lo que está dentro de este archivo). Luego podrás abrir el main-bin.exe para ver los cambios que vayas implementando.

mlibarra, 2020-05-23:
Lo siento, se borro mi comentario anterior. El gcc lo instale buscando en "https://github.com/msys2/MSYS2-packages/issues/293" instalando pacman -S mingw-w64-x86_64-toolchain

felipesanchezvaras, 2020-05-23:
@Jbijit En la imagen que muestras, indica que un posible camino es reiniciar el pc. Prueba con eso e intenta seguir con los pasos para instalar. Si esto no funciona, te sugiero desintalar todo lo que hayas instalado, reiniciar el pc y volver a intentar. Me avisas como vas.

Jbijit, 2020-05-23:
@felipesanchezvaras  muchas gracias, ahi funcionó, tuve que reiniciar el pc y después terminar de instalar todo, pero esto me aparece esto al tratar de abrir el main-bin.exe, disculpa todas las preguntas..... 

[image]

Jbijit, 2020-05-23:
@felipesanchezvaras muchas gracias, ahi funcionó, tuve que reiniciar el pc y después terminar de instalar todo, pero aún me aparece esto al tratar de abrir el main-bin.exe, disculpa todas las preguntas.....

[image]

Y tengo las carpetas en las variables de entorno.....

felipesanchezvaras, 2020-05-23:
@Jbijit Te recomiendo, que trates de compilar el programa tú (con el archivo compile.script) y que con el archivo que se genere, intentar abrirlo (o intentar abrirlo desde MSYS2). Ya que puede que estemos con versiones distintas.

CristobalMery, 2020-05-23:
Buenas tardes. Escribo puesto que el archivo compile.script no me funciona correctamente. Cada vez que lo ejecuto, me lanza lo siguiente:
[image]
Por ende, no puedo ver reflejados los cambios que realizo en el código, en el comportamiento de la interfaz.

CristobalMery, 2020-05-23:
> Lo siento, se borro mi comentario anterior. El gcc lo instale buscando en "[msys2/MSYS2-packages#293](https://github.com/msys2/MSYS2-packages/issues/293)" instalando pacman -S mingw-w64-x86_64-toolchain

> Buenas tardes. Escribo puesto que el archivo compile.script no me funciona correctamente. Cada vez que lo ejecuto, me lanza lo siguiente:
> [image]
> Por ende, no puedo ver reflejados los cambios que realizo en el código, en el comportamiento de la interfaz.

Logré solucionar mi problema realizando lo que @mlibarra mencionó previamente: instalando **pacman -S mingw-w64-x86_64-toolchain** en la consola de msys64. Creo que aquello no se menciona en el tutorial; por ende, podría ser de utilidad si alguien más tiene el mismo problema.
Gracias igualmente!

Jbijit, 2020-05-24:
@felipesanchezvaras  llego a esto y no pasa nada..... 

[image]

felipesanchezvaras, 2020-05-24:
@CristobalMery Gracias por notarlo ! Crearé una nueva Issue para indicar esta información.

@Jbijit El hecho que no aparezca "nada" significa que la compilación se realizó con éxito y sin errores. Junto con esto, se tiene que haber creado un nuevo archivo main-bin.exe, puedes fijarte que debe haber cambiado la fecha de creación de ese archivo, por lo que si haces doble click en este archivo, deberías poder ver los cambios.

Otra forma, es abrirlo directamente desde la consola de MSYS, lo que se hace con el comando
`./main-bin.exe`

gonzalezconstanza, 2020-05-24:
Hola... al intentar compilar me aparece esto: 
[image]
Pero descargué todos los packages del enunciado y tengo ambos programas en el PATH, entonces no se qué puede ser. 
gracias de antemano

Michel1297, 2020-05-27:
Hola! tengo una duda respecto al estado "GM_END". Resulta que al setear este estado desde game_logic.c no hay ninguna función en main.c que se ejecute o haga algo al respecto con esa variable y no se como hacer que el estado GM_END gatille una acción en el main.c . Considerado eso, es posible no usarlo y crear un flag  que se ejecute  desde la función show_mem_function ( que requiere estar en estado GM_SHOW) y que muestre en pantalla los mensajes de fin de partida? Traté poniendo un if(widgets->gm_state == GM_END){}  en el void main() pero no hace nada :( . Muchas gracias de antemano!

aigalvez, 2020-05-27:
Hola! Tengo dispositivo MAC y no se me permite extraer archivos ".exe", buscando en internet las soluciones que encontré es descargar emuladores de Windows para MAC lo que no me da mucha confianza. Hay alguna alternativa para extraer propuesta? O descargo simplemente los emuladores?
Gracias de antemano!

felipesanchezvaras, 2020-05-27:
@gonzalezconstanza 
Para no dejar el mensaje sin responder, me informaron que pudiste solucionar el problema.

@Michel1297 
> Hola! tengo una duda respecto al estado "GM_END". Resulta que al setear este estado desde game_logic.c no hay ninguna función en main.c que se ejecute o haga algo al respecto con esa variable y no [...]

Efectivamente, no hay ninguna función que haga algo con GM_END, ya que esto debe ser implementado. No funciona el  ponerlo en el void main() ya que esta función se ejecuta una vez hasta las lineas 

```
    gtk_widget_show(window);                
    gtk_main();
```

Que es cuando se habilita la interfaz. Todo el procesamiento es asincrono, ya que funciona con señales que se generan cuando se presiona un botón (que en ese caso llaman a las funciones que se indican) o con los timer_1 (10 ms) y timer_2 (800 ms) que se ejecutan constantemente y llaman a las funciones para hacer alguna acción. Haciendo una analogía con un microcontrolador, es como si el main() estuviera vació y todo el procesamiento fuese realizado por interrupciones de botón y de timers.

Continuando con GM_END, este puede ser tu flag que te gatilla alguna acción, por lo que la línea `if(**app_wdgts**->gm_state == GM_END){}` podría ser agregada en la función `timer_handler_1` para que la interfaz se actualice cuando se pasa al estado de juego final. (En otras palabras que cada 800 ms se revise si el juego terminó o no). (ojo que cambié el nombre de widgets a app_wdgts, dado que ese es el parametro que recibe la función `timer_handler_1` .) Este flag podría ser modificado en alguna función del game_logic.c que tendrá relación con la lógica del juego que implementes.

Como una ayuda, si deseas cambiar el mensaje que dice "Presiona START para comenzar" a "Finalizado", se hace con la línea 

```
gtk_label_set_text(GTK_LABEL(app_wdgts->w_lbl_status) , "Finalizado");
```

@aigalvez 
> Hola! Tengo dispositivo MAC y no se me permite extraer archivos ".exe", buscando en internet las soluciones que encontré es descargar emuladores de Windows para MAC lo que no me da mucha confianza. [...]

El archivo .exe es exclusivo para Windows, Ya que la interfaz fue realizada en ese Sistema Operativo originalmente. En el enunciado aparecen un pequeño tutorial de cómo hacerlo en Mac. Te sugiero que lo revises, de todas formas, copio a un ex ayudante y buen amigo que podría ayudar con esta respuesta.
Julio Cortés @loky32

diegomarusic, 2020-05-27:
Hola, [windows] ya instalé minGW (al corroborar me aparece que es la versión 5.1.0), edité los paths como se indica en el video,  instalé MSYS2 y realicé los pasos 2 y 3, (inmediatamente tras abrir msys2.64 y en la carpeta bin dentro de mingw64), y me tiró algunos errores al momento de instalar todos los archivos. 

Ahora al intentar ejecutar compile.script me aparece esto 
[image]

diegomarusic, 2020-05-27:
(más información) después de los pasos 2, 3 y pacman -S mingw-w64-x86_64-toolchain, probé ejecutando "pacman -S gcc" y me arroja el siguiente error
[image]

felipesanchezvaras, 2020-05-27:
@diegomarusic 

Prueba realizando

```
pacman -S pkgconf
```

Eso debería solucionar el problema. @CamilaTurrieta ¿Puedes corroborarme si hay algo más que hacer ?

diegomarusic, 2020-05-27:
Hola, lo hice pero no reconoce "pkgconf" :(

[image]

loky32, 2020-05-28:
> Hola! Tengo dispositivo MAC y no se me permite extraer archivos ".exe", buscando en internet las soluciones que encontré es descargar emuladores de Windows para MAC lo que no me da mucha confianza. [...]

Hola! No es necesario extraer el .exe, puedes compilar las cosas directo en Mac de la siguiente forma:
1) Abres una ventana de terminal
2) Si no tienes el gestor de paquetes homebrew instalalo con este comando:
/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"
3) Demora, pero una vez instalado brew procedemos a instalar un complemento de gcc (el compilador) con el comando: brew install pkg-config
4) Ahora instalamos las librerías gráficas con el comando: brew install gtk+3
5) Una vez esté listo todo debemos ir desde la terminal a la carpeta donde tienes los archivos, esto se hace con el comando cd en terminal, por ejemplo: cd /Users/(nombre_usuario)/Downloads/(nombre_carpeta)
6) Finalmente en la misma terminal como ya estamos dentro de la carpeta ejecutamos el script para compilar: bash complie.script

Eso debería bastar, ahora debes repetir sólamente el paso 6) cada vez que edites los archivos .c (te sugiero comentar la linea del main que incluye windows.h) y se generará el archivo main-bin sin extensión, esto para abrirlo basta arrastarlo en una terminal que ya esté en el directorio (que ya hayas ejeuctado el paso 5)) y le das a enter, se abrirá entonces el programa. Saludos y etiquétame si tienes problemas para que me llegue la notificación.

Saludos, 
JC - Colaborador SEP.

mpintob, 2020-05-28:
Hola, tengo un problema al intentar abrir el archivo main-bin.exe. Cuando intento hacerlo directamente desde el explorador de archivos, me sale el siguiente error: 
[image]

 Cuendo lo intento desde la consola de msys, no me tira ningún error, pero la interfaz no se abre:
[image]

felipesanchezvaras, 2020-05-30:
> Hola, lo hice pero no reconoce "pkgconf" :(
> 
> [image]

@diegomarusic 
Comentame si es que lograste solucionarlo. Al parecer el comando que instala directamente pkg config es 

```
pacman -S pkg-config
```

@mpintob 
Podrías tratar de trabajar en otra carpeta y no directamente en el disco C, es probable que estés teniendo problemas con los permisos de usuario.

diegomarusic, 2020-05-30:
Hola profesor, no lo he podido solucionar. Hice lo que dijo y funcionó pero me sigue arrojando un error al momento de ejecutar compile.script

intentaré desinstalar todo y volver a instalar porque creo que hice un paso de más que está produciendo el error

[image]

sebapulgar, 2020-06-02:
Hola profe, tengo mac y a la hora de ejecutar el archivo main-bin sin extensión que se genera al compilar el programa, se produce el siguiente error, el cual no entiendo ya que todas las carpetas que tengo descargadas contienen los archivos que están el github
Instalé todas las librerías varias veces y aún sigue sin funcionar
mu
[image]
chas gracias

loky32, 2020-06-02:
@sebapulgar Hola, aquí va adjunto un video para correr la aplicación en Mac OS X.
[https://youtu.be/SDoTBCME6n8](url)
Saludos,
JC - Colaborador SEP.

imgurruchaga, 2020-06-02:
Profesor @felipesanchezvaras tengo un problema relacionado con el main-bin.exe. Resulta que, al igual que otros compañeros, al correr el archivo desde msys no tengo problema pero al abrirlo directamente me tira error, que creo que es muy parecido al error de @Jbijit . Tengo windows, he instalado todo, y me corre el compile.script perfectamente. El problema de fondo es que cuando corre en msys se me queda pegado después de cierto tiempo de juego. 
[image]

felipesanchezvaras, 2020-06-02:
@imgurruchaga ¿ Lograste abrir la interfaz en algún minuto ? ¿O aún no ? Podrías intentar abrirla pero no desde MSYS2 sino que haciendo doble click sobre el archivo .exe

De todas formas, implementé una alternativa a la interfaz, utilizando un máquina virtual de Ubuntu. Si deseas utilizarlo, puedes acceder aquí https://github.com/IEE2463/classroom/issues/72#issue-629502697

imgurruchaga, 2020-06-02:
@felipesanchezvaras acabo de resolver el problema añadiendo un una tercera variable de entorno al path: C:\Program Files\Git\mingw64\bin. Efectivamente antes solo podía abrirla desde MSYS2 pero no haciendo doble click en main-bin.exe. Muchas gracias de todas formas.

jpcrisostomo, 2020-06-05:
Hola!
Al cargar la interfaz a través del terminal (Mac) tal como se hace en el video, sí se me genera el archivo main-bin, pero al momento de ejecutarlo en el terminal me arroja un error siempre. Adjunto SS.
Al crear la carpeta glade que contiene los códigos main, estos qué extensión deben tener? Los tengo en.xml (creado en Sublime text)
Gracias!

[image]

closed 2020-06-16 by felipesanchezvaras