IEE2463/classroom #65: LAB 04 - Linear Buffer

felipesanchezvaras, 2020-05-19:
Dudas relacionadas con Linear Buffer

Michel1297, 2020-05-25:
Hola, ya cree en un código aparte mi funcion generadora del  linear buffer con los n valores aleatorios de la secuencia para un juego, mi problema es que necesito acceder al contenido de los buffers en el archivo game_logic.c cuando el buffer está definido dentro del main en main.c. Estoy un poco confundido. ¿Cómo puedo "conectar" mis variables  para poder acceder a ellas (en main.c) desde game_logic.c ?

felipesanchezvaras, 2020-05-25:
@Michel1297 

Todo dependerá de cómo crees la organización del programa. Una opción es que la función generadora del buffer es encuentre en game_logic.c y no en el main.c, ya que este último archivo es el "output" del juego, ya que se encarga de controlar la interfaz y no las variables del juego que son creadas y utilizadas en game_logic.h y en game_logic.c. 

Una opción distinta, es utilizar la estructura app_widgets que controla la información del juego. Ya que con esta puedes traspasar la info de un código al otro. Intenta seguir la lógica de la generación de una secuencia se guarda en la variable `game_show` que es utilizada tanto en main.c como en game_logic.c. 

Quedamos atentos.

moquezada, 2020-06-01:
Hola!

Tengo entendido que los buffers lineales son arreglos **temporales**, pero me queda la duda de si es posible crear un *struct* para este tipo de buffer y solicitar espacio en memoria para el arreglo y todas sus componentes 🤔 Esto con el fin de utilizarlo adecuadamente tras cada iteración y una vez terminada la ronda del juego simplemente liberar la memoria

fffonceca, 2020-06-01:
Hola @moquezada. Esto es totalmente personal, ya que el los arreglos del buffer no alcanzarán a llenar totalmente la memoria RAM. Recordemos que son arreglos temporales, y si bien no son dinámicos, son bastantes cortos para este laboratorio. Por lo tanto, queda totalmente a tu criterio implementar un struct y usar funciones como malloc, etc...

rhastudillo, 2020-06-03:
Hola! El comando g_snprintf() que escribe el string en la variable game_show es la única forma de visualizar los datos ? Hay alguna forma de pasar los datos a la variable mediante un arreglo de números (directamente desde el buffer) o habrá que generar un string cada vez que que necesite cambiar app_wdgts->game_show?

CamilaTurrieta, 2020-06-03:
Hola @rhastudillo! Alegremente no es la única forma de hacerlo. No se cómo lo estas implementando tu pero te aconsejo que veas a app_wdgts->game_show como tu buffer lineal de tamaño pre-definido  y puedes ir cargandole datos como a cualquier arreglo: 

app_wdgts->game_show[1] = '2'
app_wdgts->game_show[1] = 2 + '0'

Ahí tendría dos formas de asignar el char 2 a la posición 2 del arreglo app_wdgts->game_show 

Todo va a depender de cómo lo vayas planteando.

closed 2020-06-16 by felipesanchezvaras