IEE2463/classroom #66: LAB 04 - Ring Buffer

felipesanchezvaras, 2020-05-19:
Dudas relacionadas con Ring Buffer

mlibarra, 2020-05-28:
Hola! Para este modo de juego, el puntaje máximo cómo sería? Me imagino que si hay una secuencia de 3 colores, al completarla el máximo es 3, pero luego aparece otra secuencia de 3 colores, si se completa el máximo es 6? Saludos

fffonceca, 2020-05-29:
Hola! Para responder esta issue acabo de crear una explicación improvisada para eliminar cualquier duda sobre ring buffer o la implementación de esta que pueda surgir. Con respecto a tu pregunta, recuerda que el tamaño del buffer solo puede ser cambiado **antes de compilar**, y debe ser destacado dentro del código para poder ser cambiado para su evaluación.

Lo demás se responder en la siguiente imagen:

[image]

mlibarra, 2020-05-29:
Si tengo dudas con la implementación que hice (respecto a si cumple lo pedido), puedo consultarlo en mi repositorio?

rhastudillo, 2020-06-06:
Entonces, para ver si entendí, el máximo de datos a recordar (en el último nivel) siempre será n, solo que el elemento 1 pasa a ser el 2do del original, y el último pasa a ser el nuevo ?

felipesanchezvaras, 2020-06-06:
@rhastudillo Así es !

IsabelJamett, 2020-06-06:
Hola!
Entonces solo se deben mostrar los ultimos n colores en la interfaz y la persona debe presionar solo maximo n colores? O se deben mostrar todos igualmente?

CamilaTurrieta, 2020-06-06:
Hola @IsabelJamett cuando seleccionas el modo Ring Buffer y el juego parte debería pasar lo siguiente:
1. Muestra un color.
2. Apretas correctamente el color
3. Muestra el mismo primer color y otro color.
4. Vuelves a apretar la secuencia correctamente

Así hasta alcanzar el tamaño del buffer, supongamos de tamaño n.

Cuando alcanzas ese punto y se sigue anotando correctamente. La secuencia a anotar será sólo de n elementos siempre . ¿Me explico? 

Recuerda que, cuando ya sólo se muestra secuencias de n elementos, luego de que hayas ingresado la secuencia correcta, la interfaz mostrará los últimos n-1 elementos de la secuencia anterior + un nuevo alemento generado aleatoriamente. 

Cuéntame si esto lo dejo más claro o dejo más dudas.

closed 2020-06-16 by felipesanchezvaras