IEE2463/classroom #67: LAB 04 - Linked List

felipesanchezvaras, 2020-05-19:
Dudas relacionadas con Linked List

danielauu, 2020-06-06:
Hola, estoy un poco aproblemado con la inicialización en el main de mi linked list. Definí mi estructura tipo linked list y a la estructura del juego le incorporé 2 punteros del tipo linked list. En el main, intento asignar espacio en memoria para mis linked list llamando a malloc y el código ya no compila.  😢

Michel1297, 2020-06-08:
Hola, tal como me lo propusieron @felipesanchezvaras   y @CamilaTurrieta  , el error se solucionó instanciando el nodo dentro de una función y  creándolo afuera de esta de la siguiente forma : "struct Nodo * Nodo1; " ( le quité el NULL) , y aunque siguen aparenciendo warnigs, el código funciona!

closed 2020-06-16 by felipesanchezvaras