IEE2463/classroom #69: LAB 04 - Otras dudas

felipesanchezvaras, 2020-05-19:
Espacio para responder otras dudas.

CristobalMery, 2020-05-23:
Buenos días! Escribo puesto que los cambios que le hago al código no logró verlos implementados en el archivo ejecutable. 
Hay que realizar algo adicional para que al modificar el comportamiento del archivo? 
Intenté ejecutar el archivo compile.script con MSYS2 MINGW64 pero al parecer eso tampoco funcióno.

felipesanchezvaras, 2020-05-30:
@CristobalMery 
No estoy seguro si es que habrás resuelto tu duda o no.

Para ver los cambios en el archivo ejecutable, debes primero compilar los archivos .c, para esto en MSYS debes ejecutar el archivo `compile.script` ya que este genera un nuevo main-bin.exe (es importante que al momento de compilar tengas la interfaz cerrada. Puedes ver la pregunta y respuesta de Jbijit en https://github.com/IEE2463/classroom/issues/64#issuecomment-633157749

RafaelOrmazabal, 2020-06-01:
Hola, quería saber si necesariamente tengo que cambiar la posición del puntero original (como lo hacen en el código del publicado usando ptr++;), o puedo mantener el puntero original y recorrer los buffers usando ptr+i;. Espero que se entienda, no se como explicarlo mejor sin mostrar mi código.

fffonceca, 2020-06-01:
Hola @RafaelOrmazabal . Punteros es la vía alternativa de implementación de un buffer, sin embargo, no es mandatorio. Otras formas de hacerlo son válidas mientras se cumpla lo especificado en el enunciado. (por ejemplo, usando ptr+i).

mlibarra, 2020-06-01:
Hola, quería saber si se puede utilizar el código de LinkedList visto en la ayudantía para realizar la tarea, ya que la parte de definir la estructura linkedlist, agregar un nuevo elemento y recorrer la linkedlist creo que es una forma estándar (yo por lo menos intenté realizar un código propio de otra forma y siempre me tiraba error). Obviamente citando a los ayudantes que fueron los que diseñaron el código, y adaptándolo a lo que uno necesita.

mlibarra, 2020-06-01:
Hola, quería saber si se puede utilizar el código de LinkedList visto en la ayudantía para realizar la tarea, ya que todo lo visto de linkedlist en la ayudantía creo que es una forma estándar (yo por lo menos intenté realizar un código propio de otra forma y siempre me tiraba error). Obviamente citando a los ayudantes que fueron los que diseñaron el código, y adaptándolo a lo que uno necesita.

felipesanchezvaras, 2020-06-01:
@mlibarra 
Es posible utilizar los códigos de las ayudantías (y esto corre para los futuros laboratorios). Lo importante es citar a los ayudantes que los diseñaron.

mlibarra, 2020-06-01:
Muchas gracias! Saludos

moquezada, 2020-06-01:
Hola!

Me di cuenta que aún no tengo una invitación al repositorio de este lab 🤔 ¿aún no los crean o simplemente no me ha llegado la invitación? (no descarto haberla borrado por accidente 🧙)

felipesanchezvaras, 2020-06-01:
@moquezada 
Para crear tu repositorio debes ir al link que se encuentra en el classroom.

[image]

moquezada, 2020-06-01:
gracias, entre tantos ramos con github olvidé que en este se hacía así

[image]

RafaelOrmazabal, 2020-06-02:
Hola! Se puede no ocupar ciertas variables del game_logic.h? Lo pregunto porque ya hice el linear y ring buffer y no toque el game_show[]. Me acabo de dar cuenta de su existencia :c.

felipesanchezvaras, 2020-06-02:
@RafaelOrmazabal 
La variable game_show[], así como otras que son utilizadas están de sugerencia. Mientras cumplas con especificado en el enunciado, no hay problemas con no ocuparlas.

dari0s, 2020-06-04:
Hola ! me decidí por utilizar Atom como compilador de C. Sin embargo, al momento de ejecutar el codigo main.c , el compilador me arroja el siguiente error:

##############################################
gcc-make-run: Compile Error

main.c:18:10: fatal error: gtk/gtk.h: No such file or directory
   18 | #include <gtk/gtk.h>
      |          ^~~~~~~~~~~
compilation terminated.
##############################################

Ya verifiqué que se instaló correctamente el paquete pkg utilizando $  pkg-config --modversion gtk+-3.0 (msys2 me arroja la versión).

felipesanchezvaras, 2020-06-04:
@dari0s 
Tuve un error similar, pero compilando desde Sublime Text. Intenté varias cosas pero ninguna funcionó. Lo único que me resolvió el problema fue compilar desde MSYS2. 

Por lo que editaba el código desde Sublime y luego compilaba desde MSYS2.

dari0s, 2020-06-04:
@felipesanchezvaras 
Muchas gracias por la respuesta, haré lo mismo jejej

mlibarra, 2020-06-06:
Hola, quería saber si el largo del buffer lineal es igual al del buffer circular o pueden ser distintos?

CamilaTurrieta, 2020-06-06:
Hola @mlibarra, queda a tu criterio. Si quieres pueden tener largos distintos o largos iguales. Tiene que estar parametrizado, eso es lo importante. :D

closed 2020-06-16 by felipesanchezvaras