IEE2463/classroom #70: LAB 04 - Respuestas útiles.

felipesanchezvaras, 2020-05-24:
Las siguientes son respuestas útiles para algunos problemas detectados. 
Para usuarios Windows:

> Lo siento, se borro mi comentario anterior. El gcc lo instale buscando en "[msys2/MSYS2-packages#293](https://github.com/msys2/MSYS2-packages/issues/293)" instalando pacman -S mingw-w64-x86_64-toolchain

> Buenas tardes. Escribo puesto que el archivo compile.script no me funciona correctamente. Cada vez que lo ejecuto, me lanza lo siguiente:
> [image]
> Por ende, no puedo ver reflejados los cambios que realizo en el código, en el comportamiento de la interfaz.

Logré solucionar mi problema realizando lo que @mlibarra mencionó previamente: instalando **pacman -S mingw-w64-x86_64-toolchain** en la consola de msys64. Creo que aquello no se menciona en el tutorial; por ende, podría ser de utilidad si alguien más tiene el mismo problema.
Gracias igualmente!

_Originally posted by @CristobalMery in https://github.com/IEE2463/classroom/issues/64#issuecomment-633133720_

danielaguilera, 2020-05-28:
Hola buenas! Seguí todos los pasos de la guía, me metí al path donde están los archivos del laboratorio e incluso instalé todos los paquetes indicados en el comentario anterior. Sin embargo, me sigue arrojando el mismo error:
[image]

danielaguilera, 2020-05-28:
Además me arroja el siguiente error al intentar abrir

[image]

RafaelOrmazabal, 2020-05-29:
> Además me arroja el siguiente error al intentar abrir
> 
> [image]

Tengo exactamente el mismo error

RafaelOrmazabal, 2020-05-29:
@danielaguilera Yo lo estaba corriendo desde mingw64 y me funcionaba más o menos, se demoraba mucho en imprimir. Ahora lo arregle agregando el segundo path que indico el profe, pensé que lo había hecho pero se me pasó.

fjpessot, 2020-05-30:
> Hola buenas! Seguí todos los pasos de la guía, me metí al path donde están los archivos del laboratorio e incluso instalé todos los paquetes indicados en el comentario anterior. Sin embargo, me sigue arrojando el mismo error:
> [image]

Me pasa lo mismo

felipesanchezvaras, 2020-06-02:
@fjpessot @danielaguilera 
Les comparto otra forma que podría ver si el error va por acá.

En la carpeta C:\msys64\mingw64\lib deberían chequear que pkg-config y gcc estén instalados.

[image]

Si están, el problema podría ser que no se encuentran.

En MSYS2 ejecuta la linea 
```
pkg-config --modversion gtk+-3.0
```
Esta debería repetir el error que te aparece a tí ,ya que no lo encontrará, en mi caso, obtengo esto:

[image]

Lo cual ocurre porque  PKG_CONFIG_PATH. encuentra la carpeta que contiene los archivos, esto se puede chequear haciendo 
```
echo $PKG_CONFIG_PATH
```
[image]

Que devuelve la carpeta que apunta. Es probable que esa configuración no esté, por lo que la forma de arreglarlo sería en  caso de que apunte a otra carpeta, seriá:
```
export PKG_CONFIG_PATH=/mingw64/lib/pkgconfig:/mingw64/share/pkgconfig
```
Ahora, si el problema persiste, buscando en una Issue en Github (https://github.com/rust-lang/pkg-config-rs/issues/23) alguien lo solucionó un problema similar intercambiando el orden, es decir:

```
export PKG_CONFIG_PATH=/mingw64/share/pkgconfig:/mingw64/lib/pkgconfig
```
Por lo que al repetir la línea del principio, deberías poder ver la versión:

```
  pkg-config --modversion gtk+-3.0  
```

closed 2020-06-16 by felipesanchezvaras