IEE2463/classroom #71: Problema display de 7 segmentos

CristobalMery, 2020-06-02:
Buenas noches! Escribo por un pequeño problema que me surge al utilizar el display de 7 segmentos.  Por algún motivo, los leds del dígito 1 se me encienden con una intensidad mucho menor que los dígitos 2 y 3; siendo que los 3 están conectados mediante resistencias de iguales valores. 

¿Cual puede ser el motivo? 
Descarté también que sea un problema del pin que alimenta a aquel dígito; dado que intercambié los pines que encienden cada uno de estos dígitos; y los dígito 2 y 3 se ven perfectos, pero el 1 sigue viendose con menor intensidad. 

Adjunto una foto sobre como se me ve el display:
[image]

felipesanchezvaras, 2020-06-02:
@CristobalMery Puede que sea algún desperfecto de ese dígito en particular. Pero no te preocupes, mientras sea posible ver los números será suficiente para utilizarlo en el laboratorio.

CristobalMery, 2020-06-03:
Gracias por la respuesta! Solo me gustaría aclarar una duda más.

Para el ATmega sucede que es posible percibir a plena luz de día la intensidad de este dígito; por lo que no habría problema para el cronómetro del lab 5. Sin embargo, como la MSP arroja en sus pines voltajes menores al ATmega; hace que sea casi imposible de percibir el número que este dígito muestra. Espero que esto no sea un problema a la hora de evaluar la actividad. 

Dado que este dígito solo mostraría los números 0, 1 y 2 (pues la actividad consiste en mostrar valores de 0 a 255), se me ocurrió visualizar con leds extras cual es el valor de la centena:
0 leds encendidos -> la centena vale 0
1 led encendido -> la centena vale 1
2 leds encendidos -> la centena vale 2.
¿Vale la pena realizar esto? o revisando el código bastaría para comprobar que se está efectivamente encendiendo el dígito de las centenas?

RafaelOrmazabal, 2020-06-09:
@felipesanchezvaras Hola! Quería saber si existe la posibilidad de que pueda utilizar un arduino uno para las presentaciones en vez del ATMEGA (lógicamente sin usar la librería de arduino).

sebapulgar, 2020-06-13:
Hola! cuando me puse a hacer el lab 05 me topé con que un segmento del display no prendía, para comprobarlo usé el código del lab 02 y noté que al usar el primer dígito seguía sin prenderse el el mismo segmento a diferencia de cuando usaba los otros dígitos del display (ver imágenes).
Me gustaría saber si efectivamente creen que sea un problema del display y si es así, si sigo realizando el contador de la tarea de la misma con ese problema o elimino la centecima para no utilizar el primer dígito
(quiero destacar que los del 0 al 99 se ven bien en el primer caso pero cuando cambio de dígitos pasa lo de la segunda foto)

[image]

[image]

closed 2020-06-23 by felipesanchezvaras