IEE2463/classroom #78: Botones MSP430

tito21, 2020-06-14:
Hola, 
Cuando intento usar los botones de la MSP430 para cambiar los LEDs estos cambian cuando comienzo a acercar la mano al botón ANTES de presionarlo. Es como si la estática de mi dedo interfiere con el pin y lo cambia de valor antes. ¿Es normal este comportamiento? ¿Accidentalmente queme la tarjeta? ¿Existe una solución por software?

IsabelJamett, 2020-06-14:
Hola, tengo el mismo problema que @tito21, estoy usando interrupciones de pin para los botones, ¿que podria ser?

CamilaTurrieta, 2020-06-14:
Hola, @IsabelJamett y @tito21, vemos, si con un pequeño roce cambia, entonces, sí es normal. La diferencia es que la msp no tiene un circuito de debounce por hardware, entonces, el botón se vuelve loco con un pequeño toque. Por lo que hace necesario hacer un buen debounce por software.

CamilaTurrieta, 2020-06-14:
Otra cosa importante, son los resistores de pull up. Son importantes para evitar que los nodos queden flotando sin definición previa.  Eso también puede estar afectando en ese comportamiento loco que describen.

tito21, 2020-06-14:
El problema era que no tenia activados las resistencias de pull-up. Había asumido que la tarjeta tenía una resistencia soldada con el botón. Hay que prenderla por software.

CamilaTurrieta, 2020-06-14:
Me alegro que lo hayas solucionado @tito21! Muchas gracias por comentar la solución.

closed 2020-06-23 by felipesanchezvaras