IEE2463/classroom #80: RealTerm

mlibarra, 2020-06-16:
Hola! Cuando muevo el potenciometro y pido su valor desde RealTerm, este no se actualiza a noser que apague el puerto desde la consola y lo encienda de nuevo. Ya he revisado el código varias veces y no sabría que está mal. Alguien tiene alguna idea de qué puede ser?

fffonceca, 2020-06-17:
Prácticamente cuando cortas el puerto desde la consola y lo vuelves a conectar, reinicias el programa. Lo que esperamos en el laboratorio es que puedas mandar un mensaje desde la consola a la placa con tal de pedir el voltaje en ese instante **sin tener que apagar la consola forzadamente**. Espero que te guíe @mlibarra Saludos!

mlibarra, 2020-06-17:
@fffonceca Sii si entiendo que no debo reiniciar, sé que está mal, por eso pregunto, porque ya he revisado hartas veces el código y no sé que tengo malo. Según yo estoy actualizando el ADC, el valor de UART (la recepción con interrupciones), tengo variables volátiles, etc, entonces no sabría que ocurre. Es posible subir el código a mi repositorio? Saludos!

felipesanchezvaras, 2020-06-17:
@mlibarra Sube tu código al repositorio. Crea una Issue indicando el problema que tienes y después comentala añadiendo a los ayudantes de la forma @IEE2463/ayudantes-sep

mlibarra, 2020-06-18:
@felipesanchezvaras muchas gracias!

closed 2020-07-22 by felipesanchezvaras