IEE2463/classroom #81: Display LCD

CristobalMery, 2020-06-16:
Buenas tardes. Escribo porque según el enunciado del laboratorio 6 interpreto que hay que transmitir por UART los datos desde la MSP hacia el LCD. Sin embargo en todos los Datasheets y códigos de ejemplo que encuentro para utilizar el display, a este le envían los datos de forma paralela a sus 8 pins de escritura. Hay que realizarlo de esta manera?

felipesanchezvaras, 2020-06-16:
@CristobalMery Te sugiero que leas la respuestas que publicamos en otra Issue respecto de esto mismo. https://github.com/IEE2463/classroom/issues/74#issuecomment-642357046

Hay un modo en que se reducen los pines y recomendamos utilizar ese.

mlibarra, 2020-06-20:
Hola,
Para la actualización de la pantalla LCD, puede ser que parpadee al actualizarse? Por ejemplo si quiero actualizar un valor, puedo limpiarla, actualizar el valor y mostrar todo lo anterior pero con el valor actualizado? (lo que se ve como un parpadeo pero muy rápido).

CamilaTurrieta, 2020-06-21:
Hola @mlibarra, si estas actualizando el valor, puedes llegar y reemplazar el valor. Es decir, no es necesario que limpies lo que había en la pantalla, solo actualizas el valor. Si te es más cómodo puedes mandarle siempre el mismo tamaño de dato a la pantalla usanso ' ' (espacios en blanco). Así por ejemplo, mandas "1234" y luego en la actualización puede mandarse " 103". ¿Se entiende? 

Igual no es tan grave si parpadea al actualizarse. Mientras se actualice bien, con el valor correcto, todo estará perfecto. 

Saludos.

closed 2020-06-23 by felipesanchezvaras