IEE2463/classroom #82: Espacio colaborativo!

CamilaTurrieta, 2020-06-16:
Espacio destinado a compartir enlaces de páginas web, videos de youtube u otras cosas interesantes que puedan aportar en el desarrollo de los laboratorios. 👍 😄

CamilaTurrieta, 2020-06-16:
Hola, encontré esta página web que tiene los vectores de interrupción para la atmega328! La comparto por si alguien le es útil. 
http://ee-classes.usc.edu/ee459/library/documents/avr_intr_vectors/

GutiJos, 2020-06-17:
Hola a todos!
Siguiendo la motivación de @CamilaTurrieta les quiero compartir cómo yo buscaba los vectores de interrupción de la MSP430!
1. Hacer Ctrl+click en su línea **#include <msp430.h>**
2. Buscar la MSP430 que estamos utilizando->MSP430f5529, llegaremos a unas líneas como estas:
**#elif defined (__MSP430F5529__)
#include "msp430f5529.h"**
3. Ctrl+click en **#include "msp430f5529.h"**.
4. Al final de la pagina encontrarán un sección llamada **Interrupt Vectors (offset from 0xFF80)** donde pueden encontrar los vectores de interrupción!

Espero les sea útil!

CristobalMery, 2020-06-17:
Buenas noches! A continuación dejo un código que me sirvió como base para programar el display LCD:
Autor: Msminhas93
Link: https://www.instructables.com/id/16x2-LCD-interfacing-in-4-bit-mode/
Saludos!

fjpessot, 2020-06-19:
> Buenas noches! A continuación dejo un código que me sirvió como base para programar el display LCD:
> Autor: Msminhas93
> Link: https://www.instructables.com/id/16x2-LCD-interfacing-in-4-bit-mode/
> Saludos!

Que pines usaste para los 4 bits de datos? Yo adapte la librería para ocupar los pines p3.0-p3.3, pero no logro ni siquiera inicializar el lcd.

CamilaTurrieta, 2020-06-19:
Hola @fjpessot ¿como realizaste la conexión de la lcd al microcontrolador?, puede que no estés controlando bien el tema de el contraste. Eso se controla conectándo la salida de un potenciómetro al pin 3 de la pantalla (contando desde la izquierda).

fjpessot, 2020-06-19:
El error estaba en que habia cambiado varias cosas del puerto 1 al puerto 3, pero me faltó reconfigurar partes del puerto 1 que todavia estaba usando para los pines 4 5 y 6

closed 2020-06-23 by felipesanchezvaras