IEE2463/classroom #84: Monitor Serial LAB 06

danielaguilera, 2020-06-17:
Hola!

M edescargué RealTerm y puse todos los parámetros que se dijeron en la ayudantía. Sin embargo, no logro ver los mensajes que envía la MSP430 en la consola. ¿Alguna sugerencia de qué puedo estar haciendo mal?

Saludos,

Daniel

fffonceca, 2020-06-17:
@danielaguilera Hola Daniel, te daré algunos consejos para revisar y ver si tu código está bien.

- Primero, revisar la configuración UART, revisa que todo está bien establecido a un baudaje que tu conozcas, bit de parada, bir de paridad, etc...   (esto incluye claramente verificar si configuraste bien tus registros)
-Segundo, corrobora que esta configuración de UART sea la misma que la de Real Term con respecto a los valores anteriores.
-Tercero, recuerda que para ver los mensajes debes configurar la puerta y apretar "Change" en la consolta, procurando que usas el puerto correspondiente a la MSP430.
-Cuarto, en caso de usar interrupciones de UART, te recomiendo revisar bien el datasheet con tal de no estar haciendo algo mal.
Creo que esas serían mis recomendaciones, si alguno tiene otra la puede agregar también =D.

PD: Watchdog apagado claramente...

danielaguilera, 2020-06-17:
Hola @fffonceca,

Sabes que comprobé todas tus sugerencias y sigue sin mostrar nada :(. Utilicé tal cual el código de la ayudantía sin cambiar absolutamente nada.  Sería posible hacer una videollamada o algo así, para ver cuál es el problema?

GutiJos, 2020-06-17:
Hola @danielaguilera 
Complementando lo que dice mi compañero @fffonceca , te recomiendo fijarte en los siguiente.
1. Que efectivamente estés habilitando los bits de transmisión/recepción de UART.
2. Si tienes abierto RealTerm y no se logra ver nada en la terminal, te recomiendo que te fijes si el puerto USB está bien configurado al que estas conectando a tu MSP.
3. Ahora, si logras ver caracteres extraños, te recomiendo fijarte que la configuración de la terminal coincida con la configuración que realizaste en tu código UART.

Saludos.

danielaguilera, 2020-06-17:
Hola @GutiJos probé los 3 aspectos que me dijiste. Revisé la ayudantía y configuré exactamente con lo que dijo el ayudante, sin hacer ningún cambio. También revisé si el puerto es correcto y lo es. Sin embargo, me muestra unos caracteres muy extraños y no muestra el mensaje de bienvenida.

¿Qué más puedo estar haciendo mal? Ya no se me ocurre jaja

GutiJos, 2020-06-17:
@danielaguilera presiento que debe ser alguna configuración. Debe haber algo distinto en cómo estas configurando UART y la terminal de Realterm, porque si logras ver caracteres extraños es porque debes estar transmitiendo algo.

felipesanchezvaras, 2020-06-17:
@danielaguilera Complementando a los ayudantes, el MSP tiene un doble canal de comunicación. Este lo puedes ver en el administrador de dispositivos al momento de conectar. En la siguiente foto se muestra, para usar RealTerm debes seleccionar (en mi caso) COM3

[image]

Por otra parte, revisa que los pines del MSP estén de la siguiente forma, ya que puede que no esté conectado el MSP con el ezFET que permite realizar la conversión a USB. 

[image]

closed 2020-06-23 by felipesanchezvaras