IEE2463/classroom #85: [Atmega 328p] Problema con UART y monitor serial LAB 6

itserrano, 2020-06-18:
Hola!
Tengo una duda con respecto a la implementación de UART en la Atmega. Me pasa que si conecto mi placa y le hago flash al código, el sistema funciona correctamente. Sin embargo, cuando desconecto la placa y la vuelvo a conectar sin volver a cargar el código nuevamente, aparecen caracteres aleatorios en RealTerm. ¿Es necesario arreglar este error para el laboratorio? ¿O es necesario que funcione cuando se le quita la alimentación y se le vuelva a dar, sin cargar el código nuevamente? Pregunto por si a alguien más le ocurre y sabe cómo solucionarlo. Adjunto pantallazo de RealTerm mostrando la salida sin cargar el código nuevamente.
[image]

Muchas gracias!

fffonceca, 2020-06-18:
Hola @itserrano , cambias la configuración en algún punto de tu programa? Fíjate que algún registro de control (me refiero a UCSRnA y UCSRnB) no sea cambiado posterior a que mandes el mensaje. También te recomiendo, si ya estás usando interrupciones de recepción, revisar que tu ISR sea corto, ya que las interrupciones no deberían tardar mucho para no ser invasivo con el main.

GutiJos, 2020-06-18:
Hola @itserrano
Entiendo que desconectas la placa y cuando la vuelves a conectar te tira esos caracteres extraños sin haber cambiado nada en tu código.

Cuando te muestre esos caracteres extraños configura tu consola de realterm (seleciona correctamente el puerto y la configuración de UART que esta utilizando) y presiona el botón de reset de la placa. Todo esto sin volver a desconecta y conectar la placa.

Coméntanos como te va!

Saludos.

itserrano, 2020-06-19:
@fffonceca Revisé el programa y no cambio los registros de UART excepto en el inicio. La ISR creo que no es tan larga (espero).
@GutiJos configuré la consola, realicé los cambios con el botón, presiono reset pero sigue mostrando caracteres extraños.

fffonceca, 2020-06-19:
@itserrano Te aseguraste de apretar "Change" en Real term después de generar los cambios que esperabas?. Si no apretas "Change" no se verá el cambio de baudios en el RealTerm y la consola esperará datos a 57,6k de baudrate. 
[image]

itserrano, 2020-06-19:
@fffonceca Configuré RealTerm según cómo se envián los datos en la Atmega y funciona de forma correcta solamente cuando compilo y subo. Si desconecto la placa y la vuelvo a conectar se envían de forma correcta los datos pero se envían además esos caracteres extraños:
[image]

closed 2020-06-21 by CamilaTurrieta