IEE2463/classroom #86: Terminal Serial MAC

sebapulgar, 2020-06-18:
Hola muy buenas, no me queda muy claro que aplicación utilizar como terminal serial en mac ya que el realterm me parece que no esta disponible en el sistema operativo (o yo no lo encontré):(
No se si debo utilizar la misma terminal (aunque no se como) o descargar alguna otra aplicación que no he encontrado
muchas gracias

fffonceca, 2020-06-18:
Hola @sebapulgar . No soy un experto en MAC, sin embargo, he visto a algunas personas usando CoolTerm. Te envío un link del programa: https://www.macupdate.com/app/mac/31352/coolterm

sebapulgar, 2020-06-18:
@fffonceca gracias por la respuesta! , la verdad es que ya había descargado el coolTerm sin embargo no he logrado que funcione ya que probando el código realizado en la ayudantía de UART las respuestas que entrega la terminal no tienen mucho sentido.
no se si no he realizado alguna configuración, pero me preocupé de seleccionar 9600 baud, sin bit de paridad y mandar en ASCII así que no se cual es el problema

[image]

loky32, 2020-06-19:
Hola @sebapulgar, he utilizado coolTerm sin problemas en Mac OSX, si este no te resulta te sugeriría como opción mas rápida bajar la plataforma de Arduino y utilizar su serial monitor para descartar algún problema del codigo (o configuración de él). Con eso asegurate que el micro está enviando algo legible, si ves lo que esperas entonces podemos ver otras alternativas a coolTerm. Coméntanos como te va!
Saludos,
JC - Colaborador SEP.

jpcrisostomo, 2020-06-21:
Hola, sabes cómo compilar códigos en AVR (por terminal/makefile)  cuando se utilizan librerias .h y su respectivo código .c ?
Ya que al intentar compilar me arroja este error  con las funciones creadas en librerías aparte.
undefined reference to `USART_Init'
Gracias

felipesanchezvaras, 2020-06-21:
@jpcrisostomo Te sugiero que respondas lo que escribí en https://github.com/IEE2463/classroom/issues/89 .

@loky32 Respondemos en la otra Issue, debido que tiene relación con compilación.

closed 2020-06-23 by felipesanchezvaras