IEE2463/classroom #87: UART MSP

mlibarra, 2020-06-19:
Hola! Tengo dudas respecto a si el UART de los pines 3.3 y 3.4 comunican también por USB con el computador, o esto sólo ocurre con los pines 4.4 y 4.5? Además, P4.4 y P4.5 se pueden usar como UART para un micro externo, o sólo funcionan con USB? Saludos!

CamilaTurrieta, 2020-06-19:
Hola @mlibarra los pines de uart que van al usb son los 4.4 y 4.5. Estos si desconectas los conectores negros que estas haciendo la conexión en los pines señalados en la placa como Rx Tx, vas a desconectar el uart por usb y puedes conectarte a la avr.  El otro canal de uart no comunica con el usb.

mlibarra, 2020-06-19:
@CamilaTurrieta muchas gracias! Ahora entiendo porque dejaba de funcionar el usb... :D

dabalada, 2020-06-21:
Hola, la comunicación debe ser mediante interrupción del Rx o puede ser sin interrupciones?

Al hacerlo con interrupciones pierdo el primer caracter que se envía a la MSP.

Gracias

CamilaTurrieta, 2020-06-21:
Hola @dabalada, si hay que hacerlo con interrupciones de Rx.

dabalada, 2020-06-21:
Pese a que el enunciado solo lo pide e el task 1?
y saben por que podría ser que este perdiendo el primer caracter?

felipesanchezvaras, 2020-06-21:
@dabalada Lo que te debe estar sucediendo es que hay una configuración erronea en la interrupción. 

Debes tener claro que la interrupción de UART se activa cada vez que se termina la transmisión de 1 caracter, por lo que en ese instante debes guardar el caracter que te llega. Si es que estás recibiendo un arreglo de caracteres (un string), considera el uso de un buffer de recepción que vaya guardando los caracteres a medida que llegue y de alguna forma identificar el fin del string (y sea con un simbolo o un salto de línea ´\n´.

Respecto de la interrupción de recepción. Esta se solicita en ambos task:

[image]

closed 2020-06-23 by felipesanchezvaras