IEE2463/classroom #89: Compilar librerias en MAC

sebapulgar, 2020-06-21:
hola! inicialmente desarrollé un código en windows para la parte 1 de la tarea, sin embargo ya no cuento con ese computador y a la hora de compilar el código desde la terminal del mac como se enseñaba en el task 1 dice que las funciones declaradas en la libreria no existen a pesar de que están declaradas y se llama a la misma librería en el main
No se si debo compilar también las librerías por separado, sin embargo es el mismo código que desarrollé en atmel y funcionó por lo que no se cual es el error.

[image]

jpcrisostomo, 2020-06-21:
Hola, a mi me pasa exactamente lo mismo. No he podido compilar aún.
Buscando en internet vi que probablemente es porque el makefile/comandos para compilar terminal solo llaman al archivo UART.c, pero no se llaman a las librerías implement_me.h ni su código  implement_me.c

felipesanchezvaras, 2020-06-21:
@sebapulgar @jpcrisostomo pueden subir una foto de la carpeta en que se vean los archivos? Que comando están utilizando para compilar ?

jpcrisostomo, 2020-06-21:
Hola, esta es la carpeta en la que tengo mis códigos y el makefile
[image]
Y uso este makefile para compilar
[image]

sebapulgar, 2020-06-21:
@felipesanchezvaras  hola! tengo exactamente lo mismo, he tratado compilando desde el terminal con un bash y directamente con los comandos . La única diferencia es que tengo los ficheros dentro de una carpeta llamada UART
Por ahora para no atrasarme pasé los ficheros dentro de la misma documento main.c como funciones pero no sé que más hacer.

Los comandos usados son los que se encontraban en el task1

[image]
[image]

felipesanchezvaras, 2020-06-21:
@sebapulgar 

En tu caso, lo que está faltando es agregar la librería de uart para la compilación, si lo tienes en una carpeta te debería quedar algo como:

```
avr -gcc -Wall -Wextra -Wpedantic --std=gnu99 -mmcu=atmega328p -Os -o main.elf main.c USART/USART_implement_me.c
```

Las siguientes lineas, que tienen el `avr-objcopy` y `arvdude` se deben mantener iguales, ya que en la línea anterior, le estas diciendo al compilador que considere también el USART_implement_me.c

No es lo ideal tener todo en un mismo archivo, pero por temas de tiempo está bien si es seguiste avanzando manteniendolo ahí, aunque te sugiero tener listo lo demás. 

@jpcrisostomo En tu caso es algo muy similar, a tu compile: debes agregarle la ruta del USART_implement_me.c, es decir, luego del  `$(FILENAME).c` le puedes agregar `USART_implement_me.c` y eso reconocerá el archivo.

sebapulgar, 2020-06-21:
@felipesanchezvaras Muchas gracias funcionó perfecto!

closed 2020-07-03 by felipesanchezvaras