IEE2463/classroom #92: LAB 07 - Uso y creación de librerías.

felipesanchezvaras, 2020-06-23:
Espacio destinado para preguntar por el uso y creación de librerías.

mlibarra, 2020-06-25:
Hola! Existen dos carpetas con librerías de SPI, una para SD y otra para TFT, y las funciones de cada una son las mismas. Se debe llenar sólo una librería?

jffarren, 2020-06-25:
Porfavor respondan la pregunta anterior. Hay muchas carpetas USART y archivos repetidos... Estamos bastante perdidos respecto a lo que hay que hacer exactamente. Lo otro, ¿hay que hacer un proyecto independiente por carpeta? ¿O hay una forma rapida de llamar todos los archivos desde un sólo proyecto? ¡GRACIAS DE ANTEMANO! @felipesanchezvaras

felipesanchezvaras, 2020-06-25:
@mlibarra 
Las funciones SPI de la SD (utilizada en ayudantía) y TFT son levemente distintas, deben abrir la de la TFT y rellenar lo que requiere para su funcionamiento. Si copian y pegan el SPI de la SD,  **la TFT no funcionará**.

@jffarren 
Hay muchas carpetas, guiaré un poco.

En todas las carpetas encontrarán:

- [ ] .../USART/USART_implement_me.c
- [ ] .../USART/USART_implement_me.h

Son las mismas funciones de USART utilizadas en el laboratorio anterior, por lo que pueden reciclar sus códigos.

- [ ] Material I2C/I2C/I2C_implement_me.c
- [ ] Material I2C/I2C/I2C_implement_me.h

Son las funciones de I2C desarrolladas en ayudantía, por lo que deben ser rellenadas. De aquí:
- [ ] i2c_temperature.c
Es el archivo usado en ayudantía del TMP102, sin cambios. **Ojo que este no tiene decimales de precisión, ya que es truncado. Por lo que deben modificar `tmp102Read` para que entregué al menos 1 decimal de precisión.

-[ ] i2c_DAC.c
Es un prototipo de cómo debería verse un código que solo utiliza el DAC, aquí en el while(1) hay indicaciones de cómo utilizarlo y se pueden apoyar también en el [Datasheet ](https://www.sparkfun.com/products/12918) del mismo.

- [ ] Material SPI - SD / ...

Incluye todo el material **en blanco** usado en la ayudantía de SPI. Por lo que si desean probar la SD, deben rellenar estas funciones.

- [ ] Material SPI - TFT/SPI/SPI_implement_me.c
- [ ] Material SPI - TFT/SPI/SPI_implement_me.h

Son las funciones para la TFT, son **distintas** a las de la SD, por lo que deben abrirlas para poder comprender cómo usarlas y rellenarlas.

- [ ] Material SPI - TFT/display/...

Todos los archivos relacionados con la pantalla TFT.

Por otro lado, en Atmel Studio, tienes las siguientes opciones:

- Crear proyectos por separado para cada Task.
- Crear distintos main para cada Task
- Utilizar macros para compilar secciones del codigo por separado (Hay mucho material disponible, pueden [consultar aquí ](https://www.tutorialspoint.com/cprogramming/c_preprocessors.htm) cómo usarlas):

```
#define TASK_1

#ifdef TASK_1
// Si es que la linea #define TASK_1 se encuentra en el codigo, el compilador considerará esta porcion de código
printf("La macro TASK_1 esta definida\n\r);
#else 
// Si no se encuentra la linea #define TASK_1 todo lo que este dentro de este #else no sera considerado
printf("La macro TASK_1 no esta definida\n\r);
#endif 

```

Independiente de lo anterior, es importante agregar todos los archivos al proyecto en Atmel Studio, para ello, al momento de iniciar un proyecto nos encontraremos con las carpetas de esta forma

[image]

Luego al copiar a la carpeta del proyecto, sucederá que **no** serán cargadas en Atmel Studio
[image]

Por lo que para que aparezcan es necesario incorporarlas al proyecto. Esto se hace presionando el botón "Show all files"

[image]

Y haciendo click derecho sobre cada una de las carpetas e indicar "Include in project"

[image]

Lo que luego de añadir cada carpeta, se verá de esta forma:

[image]

Espero esto pueda servir para orientar un poco el trabajo, 

Saludos !

jffarren, 2020-06-25:
Muchas gracias!!

Jbijit, 2020-06-25:
@felipesanchezvaras , @IEE2463/ayudantes-sep , Hice lo que se dijo de meter todo en un proyecto y cuando ejecuto los códigos me aparece el siguiente error:

[image]

Ahora no entiendo bien porque pasa esto..... Lo he tratado de solucionar un buen rato y aun no lo logro.....

felipesanchezvaras, 2020-06-26:
@Jbijit Eso podría estar pasando porque subiste más de un archivo igual. Revisa que no hayan archivos repetidos.

Jbijit, 2020-06-26:
@felipesanchezvaras acabo de revisar y subi las 3 carpetas 1 vez.

Según entendí, eso hay que hacer o no?

felipesanchezvaras, 2020-06-26:
@Jbijit No. Tienes que subir las librerías que vas a utilizar para el proyecto solo una vez, porque sino estarás subiendo varias veces la librería de uart, varias veces la de I2C y pasarán estas múltiples definiciones.

Jbijit, 2020-06-26:
@felipesanchezvaras Deje solo la carpeta material I2C y tuve que quitar el main del archivo i2c_temperature.c para que me corriera el archivo DAC por si solo. Ahora entiendo el problema, gracias.

closed 2020-07-03 by felipesanchezvaras