IEE2463/classroom #93: LAB 07 - I2C

felipesanchezvaras, 2020-06-23:
Espacio destinado para preguntas relacionadas con I2C y con los periféricos TMP102 y MCP4725

felipesanchezvaras, 2020-06-26:
Datos extras perífecos

- MCP4725 https://www.sparkfun.com/products/12918
- TMP102 https://www.sparkfun.com/products/13314

RafaelOrmazabal, 2020-06-27:
Hola!
Tengo una pregunta, si paso de modo DAC al menu,  el output del DAC volver a 0, o puede quedarse con su último valor?

felipesanchezvaras, 2020-06-27:
@RafaelOrmazabal Puedes decidir lo que más te acomode, o tal vez dejar la opción al usuario.

jffarren, 2020-07-01:
Hola, unas dudas pequeñas sobre el gráfico. Con mi compañero estamos haciendo una versión adaptada del Task 1.1 (TMP) en donde graficamos el valor leído en un potenciómetro (ya que él no recibió un TMP). Aqui van las dudas:

- ¿El gráfico debe ser estático o puede estar en constante movimiento?  Digo esto porque por el momento tengo fijado un tiempo de muestreo y la pantalla muestra una serie de puntos correspondientes al nivel de voltaje medido (por supuesto con los niveles de precisión requeridos) que se van actualizando a medida que se lee el ADC.

- ¿De ser estático, debemos fijar una cantidad de muestras tomadas por el usuario?

Gracias de antemano!!

felipesanchezvaras, 2020-07-01:
@jffarren 
Decisión de ustedes! 
Puede ser una ventana movil (de la forma de un osciloscopio) en donde muestran por ejemplo, 20 muestras en un intervalo de 0 a 10 seg, luego otras 20 muestras de 0 a 20 segundos, etc. Asi los puntos solo se van desplazando por el tiempo.

¿A que se refieren con constante movimiento ? ¿Se refiere a que se desplaza el grafico completo ? 

Si deciden dejarlo estático, la pregunta que se deben responder es ¿Hay alguna aplicación en donde sea útil definir un set de puntos y dejarlo estático?

closed 2020-07-03 by felipesanchezvaras