IEE2463/classroom #95: LAB 07 - TFT y SD

felipesanchezvaras, 2020-06-23:
Espacio destinado a la utilización de la TFT y SD,

RafaelOrmazabal, 2020-06-24:
Hola! Con mi pareja logramos que nos funcionara la TFT, pero el único problema es que hay ciertos caracteres que no es capaz de mostrar. Algunos de estos son la "E" (la "e" si funciona), la "P"  (la "p" si funciona) o cualquier número.

fffonceca, 2020-06-24:
Hola @RafaelOrmazabal . Es lamentable que no te funcionen esas letras :/. Podrías tratar de revisar la librería y buscar qué podría estar mal, o qué está interfiriendo en tu código. Buscar la funcionalidad de crear 1 letra y replicarla también es una opción, basta con un par de iteraciones.
En esos casos particulares, lo que hacía es generar una letra "a mano". Es un poco costoso en RAM y tarda un poco de tiempo dibujarlo...
Como en las librerías están incluidos los colores, podríamos definir de esta forma:
```
#define C0 ST7735_BLACK
#define  C1 ST7735_CYAN 
```
Y así podríamos generar un arreglo que describa una letra o número, en mi caso escribiré un "uno":
```
static unsigned int uno[] =
{       C0, C0, C0, C0, C0, C0, C0, C0,
	C0, C0, C0, C0, C1, C0, C0, C0,
	C0, C0, C0, C1, C1, C0, C0, C0,
	C0, C0, C1, C0, C1, C0, C0, C0,
	C0, C0, C0, C0, C1, C0, C0, C0,
	C0, C0, C0, C0, C1, C0, C0, C0,
	C0, C0, C0, C0, C1, C0, C0, C0,
	C0, C0, C1, C1, C1, C1, C1, C0
};
```
Me cuentas como te va, y si resuelves tu problema sería bueno que lo compartieras en caso de que a otro le pase. Saludos!

CristobalMery, 2020-06-24:
Buenos días. Tengo exactamente el mismo problema que Rafael. Las funciones drawChar y drawtext no logran escribir en el display casi ninguna letra mayuscula;  y ningún número. Voy a intentar programar cada numero por si solo.

CamilaTurrieta, 2020-06-24:
Hola @CristobalMery y @RafaelOrmazabal ¿Cómo están llamando a los números?

felipesanchezvaras, 2020-06-25:
@CristobalMery y @RafaelOrmazabal 
Subimos una nueva versión de la librería. Cambiaron los archivos 

```
display/graphic_shapes.c
display/graphic_shapes.h
display/ST7735_commands.h
displaytest.c
```

Ahora están operativos los números y las letras mayúsculas.

Créditos a @CamilaTurrieta que detectó y corrigió el error.

fjpessot, 2020-06-26:
El codigo de prueba de la tft deberia funcionar sin modificar el codigo? A mi y a mi compañero no nos ha funcionado y queriamos consultar donde han tenido que modificar si hantenido que hacerlo.

felipesanchezvaras, 2020-06-26:
@fjpessot el código de los ejemplos no funciona sin modificar, hay una pequeña parte que ustedes deben rellenar. Como guía, lean lo que aparece en el main de displaytest.c y fijarse bien en las conexiones.

mpintob, 2020-06-28:
Hola, tenemos un problema al probar el la pantalla, ya que la función SPI_Master_transmit_char solo se ejecuta las dos primeras veces y despues el programa se queda pegado. En algunas ocasiones el programa empieza a correr  y muestra hasta la mitad el color rojo, luego se queda pegada. 
[image]

fffonceca, 2020-06-28:
@mpintob Sospecho que podría ser. Probablemente tu while o for loop que generas para recorrer toda la pantalla esté mal definida. Recuerda que las dimensiones de la pantalla son las siguientes (lo encuentras en la librería graphic_shapes.h):
```
#define TFT_HEIGHT 128
#define TFT_WIDTH 160
```
Además, procura que efectivamente el largo sea el largo y el ancho el ancho. Me refiero a que no estés iterando en dimensiones equivocadas. Cuéntame si solucionas el problema. Saludos y ánimo!

closed 2020-07-03 by felipesanchezvaras