IEE2463/classroom #96: LAB 07 - Menú de usuario y máquinas de estado

felipesanchezvaras, 2020-06-23:
Espacio destinado al uso e implementación de un menú de usuario y máquinas de estado.

CristobalMery, 2020-06-27:
Buenas tardes; tengo una duda respecto del Task 3: "Mejoración del Sistema".

No entiendo muy bien qué animaciones son las que "se encuentran en la interfaz". Podría ser por ejemplo que al moverme entre las distintas pantallas; muestre una de las animaciones que se entregan en el código displaytest.c?

fffonceca, 2020-06-27:
Las animaciones deben ser una ayuda o información que ayude a ver el estado del sistema actual. Por ejemplo, el mini reloj que indica la hora para transmitir el tiempo, o implementar una animación de bienvenida para el inicio. Deben ser creados por ustedes por lo mismo. Pueden basarse en la librería displaytest, pero haciendo los cambios oportunos para que se cumple la Task3.

sebapulgar, 2020-06-28:
Hola, no entiendo bien el funcionamiento del reloj que debemos hacer ya que dice que al iniciar el sistema debemos enviar la hora, pero no me queda claro si esto debe ser un proceso automático o realizar una pregunta por la terminal donde debemos ingresar la hora actual
Muchas gracias!

felipesanchezvaras, 2020-06-30:
@sebapulgar 
Puedes decidir que es lo que encuentras más cómodo. 
Realizar el proceso automático es posible, pero más complejo. Ya que necesitas de algún programa que detecte la solicitud del sistema al momento de iniciarlo y ser capaz de enviarlo de vuelva al microcontrolador.

El proceso puede ser simplemente enviar la pregunta al terminal y escribir la  hora actual y enviarla al micro. Incluso, puedes prescindir de la pregunta y solo enviarla, dejando este proceso que sea la primera acción del microcontrolador.  (Así ahorras un poco de memoria).

closed 2020-07-03 by felipesanchezvaras