IEE2463/classroom #97: LAB 07 - Compilación y carga de programas

felipesanchezvaras, 2020-06-23:
Este laboratorio, hay que optimizar procesos porque podría ocurrir que se limite la memoria del microcontrolador. Por lo que hay que tener cuidado con la compilación. Este espacio está destinado a esto y a cómo compilar los programas con muchas librerías, tanto en Atmel Studio, como al usar avr-gcc en Linux y MAC.

felipesanchezvaras, 2020-06-25:
Los usuarios de MAC y LINUX, que deseen compilar los archivos desde la consola de comandos, deben hacerlo de la siguiente forma

## I2C

### Para el DAC
```
avr-gcc -Wall -Wextra -Wpedantic --std=gnu99 -mmcu=atmega328p -Os -o i2c_TMP.elf i2c_TMP.c USART/USART_implement_me.c I2C/I2C_implement_me.c
avr-objcopy -j .text -j .data -O ihex i2c_TMP.elf i2c_TMP.hex
avrdude -p atmega328p -F -c stk500v1 -P /dev/ttyACM0 -b57600 -D -V -U flash:w:i2c_TMP.hex:i
```

### Para el DAC
```
avr-gcc -Wall -Wextra -Wpedantic --std=gnu99 -mmcu=atmega328p -Os -o i2c_DAC.elf i2c_DAC.c USART/USART_implement_me.c I2C/I2C_implement_me.c
avr-objcopy -j .text -j .data -O ihex i2c_DAC.elf i2c_DAC.hex
avrdude -p atmega328p -F -c stk500v1 -P /dev/ttyACM0 -b57600 -D -V -U flash:w:i2c_DAC.hex:i
```

## SPI

### Para la SD

```
avr-gcc -Wall -Wextra -Wpedantic --std=gnu99 -mmcu=atmega328p -Os -o SDcardtest.elf USART/USART_implement_me.c SPI/SPI_implement_me.c PetitFatFilesystem/avr_mmcp.c PetitFatFilesystem/pff.c SDcardtest.c
```

### Para la TFT 
```
avr-gcc -Wall -Wextra -Wpedantic --std=gnu99 -mmcu=atmega328p -Os -o displaytest.elf SPI/SPI_implement_me.c USART/USART_implement_me.c display/ST7735_commands.c display/graphic_shapes.c display/testroutines.c displaytest.c
```

### Comun a ambos
```
avr-objcopy -j .text -j .data -O ihex filename.elf filename.hex
avrdude -p atmega328p -c stk500v1 -P /dev/ttyACM0 -b57600 -D -V -U flash:w:filename.hex:i
```

JMiguelCathalifaud, 2020-06-29:
Hola profe! Estamos teniendo problemas de memoria y no sabemos cómo optimizar más cosas, no podemos borrar más líneas. Por ejemplo, si agregamos un comando para transmitir por UART alguna frase, el programa deja de compilar y Atmel Studio indica que se está excediendo la capacidad de memoria. Tiene alguna recomendación para mejorarlo?
Gracias!

diegognaez, 2020-06-29:
Hola, me esta pasando algo similar, al agregarle lineas de uart transmit me deja de funcionar el programa correctamente. Tengo que tener un numero reducido de estas lineas para que funcione, como se podría solucionar esto?

CamilaTurrieta, 2020-06-29:
Hola @diegognaez y @JMiguelCathalifaud, probablemente si se puede optimizar más. Recuerden que las lineas de UART prácticamente no las necesitan, ya que solo van a tener que recibir. Lo otro que pueden hacer es eliminar funciones que no ocupen en las librería de los dibujos (graphic shapes). O lo otro es que puede que sus animaciones sean efectivamente demasiado complejas. Esas serían algunas ideas... no se si otros de los @IEE2463/ayudantes-sep tengan más ideas.

IsabelJamett, 2020-06-29:
Hola, siguiendo la misma pregunta, es necesario hacer uart transmit de las temperaturas? O basta con que aparezcan en el gráfico? Para reducir la cantidad de uart transmit que realizo.

CamilaTurrieta, 2020-06-29:
@IsabelJamett correcto! la idea es que todo se vea en la TFT. Entonces, esas líneas de UART estarían demás por ejemplo.

fffonceca, 2020-06-29:
Holap@diegognaez y @JMiguelCathalifaud. Para complementar lo que dice Camila, eliminar material que no usan es indispensable para optimizar código. También el uso de variables y arreglos, que usan mucha RAM y podrían ser acortadas o algunas son innecesarias. Los típicos arreglos de las figuras y animaciones siempre ocupan mucha RAM, aquí pueden trabajar modificando funciones o viendo bien cuál podría ser una mejora dentro de su optimización.

JMiguelCathalifaud, 2020-06-30:
Logramos solucionar el problema de la memoria eliminando cosas que no estaban siendo utilizadas. Ahora se nos presentó otro problema, esta vez con la interrupción de UART. Si la interrupción está comentada, el código corre perfectamente. Si está descomentada, genera problemas en la TFT y termina mostrando carácteres extraños y sin sentido. @fffonceca alguna idea de qué podría ayudar? Estamos bastante seguros de que la interrupción de UART está bien definida y configurada.

Saludos

fffonceca, 2020-06-30:
Hola @JMiguelCathalifaud, podrías partir viendo las configuraciones, tanto de la consola, como las de tu microcontrollador en términos de baudrate, bit de parada, paridad y data bits. Lo segundo es correr el código de la UART con interrupción en solitario. Es decir, verificar que el código por si solo funciona correctamente, y posteriormente integrarlo. Esto es mucho más rápido para verificar errores. Me cuentas como te va.

JMiguelCathalifaud, 2020-07-01:
@fffonceca El código de de UART con interrupción corre bien al probarlo de forma independiente. El problema es que al incluirlo con las demás funciones y el resto del programa, ocurren los errores que comentaba anteriormente.

Otro error que me ocurre frecuentemente, es que el código corre (teniendo la interrupción de UART comentada) bastante bien, pero luego de un rato (pocos segundos) se resetea. Es como que hubiera algo gatillando un RST. No logro encontrar la solución para el tema de la interrupción de UART ni del reseteo del programa, alguna idea?

Saludos y gracias!

closed 2020-07-03 by felipesanchezvaras