IEE2463/classroom #98: LAB 07 - Timers y Animaciones

felipesanchezvaras, 2020-06-23:
Espacio destinado a consultar sobre animaciones y cómo usarlas.

mlibarra, 2020-06-28:
Hola! En la parte de animar el sistema, se pide "una animación que se encuentre en la interfaz". Se refieren a alguna animación cualquiera, que se vea en la transición entre una pantalla y otra?
Por ejemplo, si cambio de pantalla usando la función fillscreen, cuenta como animación? O debe ser alguna animación mostrando por ejemplos, circulos, triángulos, etc. Gracias de antemano! :D

felipesanchezvaras, 2020-06-28:
@mlibarra 
La animación se puede ver en la pantalla de menú. La animación de transición es una de las opciones que se piden, ya que pedimos un total de 3.

Por ejemplo, aquella que dice

[image]

Puede ser considerada como "una animación que se encuentre en la interfaz.

mlibarra, 2020-06-28:
@felipesanchezvaras Ya entendí! Muchas gracias

mlibarra, 2020-06-29:
Respecto al salvapantallas, si se realiza el reloj analógico, es necesario además incluir otra animación en el salvapantalla? Ya que el reloj ocupa toda la pantalla practicamente.

IsabelJamett, 2020-06-29:
Hola, una animación de bienvenida antes de pasar al menu cuenta como "una animación que se encuentre en la interfaz"? Aun no entiendo a que se refiere. Gracias

CamilaTurrieta, 2020-06-29:
@IsabelJamett es exactamente eso. Una animación que se construya a partir de las funciones que tienes en la librería graphic shapes. Puede ser algo muy simple, por ejemplo 3 círculos rebotando y la palabra "Hola!" acompañando (el rebotando puede ser como los puntitos se mueven una vez para arriba, para abajo, para arriba y pase al menu principal). Por ejemplo, imagina que estas imágenes se superponen rápido. 

- Imagen 1: 
[image]

- Imagen 2: 
[image]

Si las pusiera como gif se verían los puntitos rebotando. Algo así, la idea es que por ejemplo cambies solo la posición de los círculos en la TFT para que sea más óptimo. Se me ocurre como primero dibujas los primeros circuitos, luego los pones en blanco esos mismos círculos (para borrarlos) y luego los dibujas en la nueva posición.  ¿Me explico?

CamilaTurrieta, 2020-06-29:
@mlibarra A mi criterio, decisión de ustedes. En mi opinión que aparezca el reloj analógico como salva pantallas (es decir, si en el sistema no se esta haciendo nada durante un rato, se pase a mostrar el reloj analógico, sería bonito). Favor @felipesanchezvaras corroborar lo que digo.

felipesanchezvaras, 2020-06-29:
@mlibarra @CamilaTurrieta Me parece bien. Crear un reloj analógico ya cumple con mostrar una animación, ya que deberán realizar el movimiento de las manecillas.

imgurruchaga, 2020-06-30:
Hola! 
A partir de lo comentado previamente, sigo teniendo cierta duda con respecto al salvapantallas. Entiendo que el reloj digital implica asociarle una animación de tiempo, pero en el enunciado se añade otro punto después diciendo: "Incorpore alguna animación que esté en la pantalla...". ¿Eso implica otra animación más, o con la del reloj es suficiente? Gracias

felipesanchezvaras, 2020-06-30:
@imgurruchaga 
Lo ilustraré con un ejemplo. 

En el caso del reloj analógico, la animación ya está implementada debido a las caracteristicas de este reloj, que es mover las manecillas para incar la hora. Un ejemplo puedes verlo en el [siguiente video ](https://www.youtube.com/watch?v=3gFMphirZaE&list=PLbjw-LLSuEIA9wmM1m0XtXoWg0lEzEEWS&index=10&t=0s)

En el caso del Reloj Digital, si lo incluyes y no se hace nada más, tendrías algo como:

[image]

Que es utilizar la función drawText con un tamaño mayor. Por lo que no estás incluyendo ninguna animación extra. Sin embargo, si a esta le agregas un cuadrado que se va completando al ir avanzando el tiempo:

[image]

Sí estás cumpliendo lo que se pide de agregar otra animación a la pantalla. Este es un ejemplo de lo que puedes hacer, pero la creatividad e idea, queda a decisión tuya.

mlibarra, 2020-06-30:
Me surgió una duda de mi implementación... "Un cuadrado que se va moviendo al ir escogiendo alguna de las opciones." se considera una animación que se encuentre en la interfaz?

felipesanchezvaras, 2020-06-30:
@mlibarra Así es ! Ojo que en ese Task (asumiendo que te refieres al 3) Se piden al menos 3 animaciones simples.

closed 2020-07-03 by felipesanchezvaras