IEE2463/classroom #102: LAB 08 Entrega (08 de julio)

felipesanchezvaras, 2020-07-03:
Hola! 

Ante las preguntas de algunos, para la entrega nro 1 del proyecto (LAB 08) consideraremos será necesario presentar el juego con jugabilidad mínima, esto significa:

- Al menos ser capaz de tener un control de su "personaje" (varía según juego). Esto puede ser con botones, UART, potenciómetro, etc.
- Versión sin pantalla inicial.
- Sin niveles de juego (al menos un nivel).
- Animación simple y sin exceso de detalles.
- No es necesario llevar control de puntaje

El objetivo de esta entrega es que puedan recibir feedback de sus ayudantes y tener una estructura mental de cómo desarrollar el proyecto. Deberán informar cuáles serán sus pasos a seguir (no es necesario un informe escrito, pero lo deben tener claro). Pueden usar todo el material de sus kits, si desean incorporar material extra, pueden hacerlo.

Crearemos una Issue para cada juego en donde pueden ir preguntando de la implementación y sus avances.

A continuación pueden volver a descargar el enunciado. 

[SEP 2020-1 Proyecto Semestral.pdf](https://github.com/IEE2463/classroom/files/4867326/SEP.2020-1.Proyecto.Semestral.pdf)

JMiguelCathalifaud, 2020-07-03:
Hola! Al aceptar crear el repositorio para el Lab08, me aparece el siguiente error.
 
"Starter code template repository was not found. The repository might be deleted, or the organization that owns the repository has restrictions on third-party access. (Resource could not be found on github.com)"

Se puede ver un pantallazo de él en el link: https://gyazo.com/682c90238f0b25ba8abb47602d0e40b1

Alguna idea de cómo solucionarlo?
Gracias!

felipesanchezvaras, 2020-07-03:
@JMiguelCathalifaud ¿Puedes volver a intentar? Cambié una pequeña configuración a la hora de importar un repositorio de "Template"

JMiguelCathalifaud, 2020-07-03:
Ahi si se pudo! Gracias!

Jbijit, 2020-07-05:
@felipesanchezvaras, @IEE2463/ayudantes-sep 

Tratando de crear el repositorio me aparece esto y no me deja crearlo.

[image]

Quedo atento a sus comentarios, para poder crearlo y subir por mientras lo que ya llevo. 

Saludos,
Joaquín Bijit

felipesanchezvaras, 2020-07-05:
@Jbijit Ahora debería estar creado, volví a inicializar la copia del repositorio de Template. Me avisan si es que tienen acceso!

Jbijit, 2020-07-05:
Ahora si, gracias!

mlibarra, 2020-07-07:
@felipesanchezvaras Hola, me aparece el mismo error de Importing started code cuando intento crear el repositorio :(

felipesanchezvaras, 2020-07-07:
@mlibarra Ahora está creado :) deberían poder ocuparlo.

mlibarra, 2020-07-07:
@felipesanchezvaras Ahora si! gracias!

closed 2020-07-22 by felipesanchezvaras