IEE2463/classroom #108: Proyecto: Nibbles

felipesanchezvaras, 2020-07-03:
Avances y dudas relacionadas con Nibbles

mlibarra, 2020-07-06:
Hola! Para la implementación, los objetos que come la serpiente quedan a decisión nuestra o deben ser números necesariamente? Saludos!

felipesanchezvaras, 2020-07-06:
@mlibarra Hola! 
Si bien el juego orignalmente ocupa números debido a que está hecho en tamaño ascii, podrían ser variados para tener una mayor movibilidad en la pantalla siempre y cuando se mantenga una similitud con la lógica del juego (y no se transforme en un Snake).

danielauu, 2020-07-06:
Hola! Tenemos un problema para dibujar la comida. Estamos usando la función rand para generar posiciones aleatorias, sin embargo hay algunas comidas que se dibujan y otras no. Ya nos aseguramos que el número caiga dentro de la pantalla con la función resto. En verdad no entendemos por qué no funciona.

fffonceca, 2020-07-06:
Hola @danielauu , corroboraste que no apareciera en el obstáculo o en los márgenes de la misma? También que no apareciera en la misma serpiente?. Cómo definiste tu función rand() o qué librería usaste para usarlo?.

danielauu, 2020-07-06:
Hola Foca! era un error super tonto con las posiciones (setear x,x en vez de x,y). Gracias en todo caso por la preocupación 😄

mlibarra, 2020-07-07:
Hola! Alguien sabe cuánto debe crecer la serpiente cada vez que come? He visto videos y al parecer cada vez que come crece más que la vez anterior, pero no sé si está relacionado a algo o es libre. Saludos!

fffonceca, 2020-07-07:
@mlibarra Esto queda totalmente a criterio tuyo, siempre y cuando se vea el crecimiento UwU.

closed 2020-07-22 by felipesanchezvaras