IEE2463/classroom #110: problema con funciones que retorna un arry

chineseJack564, 2020-08-26:
Hola, tengo una duda sobre el uso de funciones que involucra un arry.
Por ejemplo, yo quiero definir una funcion que me retorna un char arry[a][b], tengo entendido que tengo que definir la funcion como char* function(), pero no se como logro que la funcion me retorna lo que necesito. 
Tambien tengo problema cuando una funcion recibe un arry de ese tipo, en general , me confunde mucho el uso de puntero y el contenido en si cuando se trata de un arry multidimencional. Si me pueden ayudar en este tema seria demasiado bacan, muchas gracias!!!

RafaelOrmazabal, 2020-08-26:
Hola! Mi recomendación es implementar funciones que sean void (no retornan nada), pero que trabajan sobre punteros, en especial cuando se trabaja con arrays. Dentro de las funciones puedes cambiar el contenido de los distintos elementos del array al que apunta el puntero usando el operador *. Luego, dado que estas operando sobre el contenido de direcciones de memoria, puedes acceder a esa información fuera de la función usando el mismo puntero. Ahora bien, esto depende de las preferencias de cada uno.  Te dejo un ejemplo de lo que me refiero:

void cambiar_elemento(int* pointer,int largo_lista){
    int i;
    for (i=0;i<=largo_lista;i++){
        *(pointer+i)=i;
    }
}

chineseJack564, 2020-08-26:
En este caso si la funcion recibe un puntero que tiene la dirrecion de una tabla[a][b],
como en la misma funcion cambio el contenido del posicion tabla[i][j] por ejemplo?
Se realiza con operacion de puntero? Ya que tenia entendido que el puntero solo apunta al primer elemento de la tabla o no?
O puedo como decir que tabla1[a][b] = *pointer_tabla y luego trabajar con tabla1?

RafaelOrmazabal, 2020-08-26:
En primer lugar, recuerda publicar los issues en el repositorio correcto (esta issue está en 2020-1). Si bien el puntero generalmente lo definimos a partir del primer elemento del arreglo, cuando hago *(pointer+i), estoy viendo el elemento i del arreglo. Ahora bien, para el caso de un arreglo de dos dimensiones, puedes que usar un arreglo de punteros (int* punteros[5]), y donde cada uno apunte a un elemento de las sublistas.